dusty-nv/jetson-inference #117: new obj-detect model error code 11

JiangWeixian, 2017-08-01:
```
Setting up coverage_loss
Top shape: (1)
with loss weight 1
Memory required for data: 3819636872
Creating layer cluster
```

[image]

JiangWeixian, 2017-08-01:
```
*** Aborted at 1501570813 (unix time) try "date -d @1501570813" if you are using GNU date ***
PC: @     0x7f06e90f6873 std::_Hashtable<>::clear()
*** SIGSEGV (@0x9) received by PID 4780 (TID 0x7f088ce8fac0) from PID 9; stack trace: ***
@     0x7f088ab3a4b0 (unknown)
@     0x7f06e90f6873 std::_Hashtable<>::clear()
@     0x7f06e90e8346 google::protobuf::DescriptorPool::FindFileByName()
@     0x7f06e90c6ac8 google::protobuf::python::cdescriptor_pool::AddSerializedFile()
@     0x7f088b7727d0 PyEval_EvalFrameEx
@     0x7f088b89b01c PyEval_EvalCodeEx
@     0x7f088b7f13dd (unknown)
@     0x7f088b7c41e3 PyObject_Call
@     0x7f088b7e4ae5 (unknown)
@     0x7f088b77b123 (unknown)
@     0x7f088b7c41e3 PyObject_Call
@     0x7f088b76f13c PyEval_EvalFrameEx
@     0x7f088b89b01c PyEval_EvalCodeEx
@     0x7f088b769b89 PyEval_EvalCode
@     0x7f088b7fe1b4 PyImport_ExecCodeModuleEx
@     0x7f088b7feb8f (unknown)
@     0x7f088b800300 (unknown)
@     0x7f088b8005c8 (unknown)
@     0x7f088b8016db PyImport_ImportModuleLevel
@     0x7f088b778698 (unknown)
@     0x7f088b7c41e3 PyObject_Call
@     0x7f088b89a447 PyEval_CallObjectWithKeywords
@     0x7f088b76d5c6 PyEval_EvalFrameEx
@     0x7f088b89b01c PyEval_EvalCodeEx
@     0x7f088b769b89 PyEval_EvalCode
@     0x7f088b7fe1b4 PyImport_ExecCodeModuleEx
@     0x7f088b7feb8f (unknown)
@     0x7f088b800300 (unknown)
@     0x7f088b8005c8 (unknown)
@     0x7f088b8016db PyImport_ImportModuleLevel
@     0x7f088b778698 (unknown)
@     0x7f088b7c41e3 PyObject_Call
```

here is my log file

JiangWeixian, 2017-08-01:
i use `pyenv` to use python2.7.1
IN global python, `import cv2` is working, but not in `pyenv`
do i need install opencv in pyenv?

AndrewVGr, 2017-08-02:
Same situation.

I use no Python environment management tools. Pure 2.7.1.

import cv2 is working

Trace:

`
*** Aborted at 1501669588 (unix time) try "date -d @1501669588" if you are using GNU date ***
PC: @     0x7f28df818b73 std::_Hashtable<>::clear()
*** SIGSEGV (@0x9) received by PID 1619 (TID 0x7f29109eef40) from PID 9; stack trace: ***
@     0x7f290d5a14b0 (unknown)
@     0x7f28df818b73 std::_Hashtable<>::clear()
@     0x7f28df809ca6 google::protobuf::DescriptorPool::FindFileByName()
@     0x7f28df7e6dc8 google::protobuf::python::cdescriptor_pool::AddSerializedFile()
@     0x7f290dc0b7d0 PyEval_EvalFrameEx
@     0x7f290dd3401c PyEval_EvalCodeEx
@     0x7f290dc8a3dd (unknown)
@     0x7f290dc5d1e3 PyObject_Call
@     0x7f290dc7dae5 (unknown)
@     0x7f290dc14123 (unknown)
@     0x7f290dc5d1e3 PyObject_Call
@     0x7f290dc0813c PyEval_EvalFrameEx
@     0x7f290dd3401c PyEval_EvalCodeEx
@     0x7f290dc02b89 PyEval_EvalCode
@     0x7f290dc971b4 PyImport_ExecCodeModuleEx
@     0x7f290dc97b8f (unknown)
@     0x7f290dc99300 (unknown)
@     0x7f290dc995c8 (unknown)
@     0x7f290dc9a6db PyImport_ImportModuleLevel
@     0x7f290dc11698 (unknown)
@     0x7f290dc5d1e3 PyObject_Call
@     0x7f290dd33447 PyEval_CallObjectWithKeywords
@     0x7f290dc065c6 PyEval_EvalFrameEx
@     0x7f290dd3401c PyEval_EvalCodeEx
@     0x7f290dc02b89 PyEval_EvalCode
@     0x7f290dc971b4 PyImport_ExecCodeModuleEx
@     0x7f290dc97b8f (unknown)
@     0x7f290dc99300 (unknown)
@     0x7f290dc995c8 (unknown)
@     0x7f290dc9a6db PyImport_ImportModuleLevel
@     0x7f290dc11698 (unknown)
@     0x7f290dc5d1e3 PyObject_Call
`

closed 2017-08-05 by JiangWeixian

YoungSharp, 2017-12-19:
I am in the same situation.

Did you guys solved this?   Or how did you solve it ?