dusty-nv/jetson-inference #121: how to modify RPROIFused layer in TensorRT?

zimenglan-sysu-512, 2017-08-03:
Recently I try to use TensorRT to run faster-rcnn based on PVANet by modifying the example code in `TensorRT-2.1.2/samples/sampleFasterRCNN/sampleFasterRCNN.cpp`.  But I found that the layer of generating proposals/rois is `RPROIFused layer `, which I can't find where it is. I need to modify this layer so that its' input `im_info` can be changed to be 6 dimension instead of 3.

twmht, 2017-08-10:
@zimenglan-sysu-512 

I am working on a similar work.

PVANet can not be built in TensorRT due to it uses group parameter in the deconvolution layer.

Did you remove the group parameter?

twmht, 2017-08-10:
btw,

the implementation of RPROIFused in TensorRT has been hided, the source code is not exposed.

zimenglan-sysu-512, 2017-08-10:
Hi @twmht ,
I guess TensorRT can support `Deconv` ops.

zimenglan-sysu-512, 2017-08-10:
Hi @twmht,
`RPROIFused` is hidden, so I can't change `im_info` to support pvanet.

twmht, 2017-08-10:
@zimenglan-sysu-512 

yup. tensorRT supports `Deconv`.

but it does not support `group` parameter in `Deconv`

I have tried to build PVANet in TensorRT, but it throws the exception that it does not support `group` in deconvolution layer.

zimenglan-sysu-512, 2017-08-10:
Hi @twmht,
It's out of my knowledge how to remove the `group` parameter using TensorRT.

Btw, can you share you how to build PVANet in TensorRT?

closed 2017-09-26 by dusty-nv

twmht, 2017-09-29:
@dusty-nv 

a question about faster rcnn, here is the input shape of faster-rcnn in TensorRT

```
input: "data"
input_shape {
  dim: 2
  dim: 3
  dim: 375
  dim: 500
}
```

My question is, can we change the input dim after `CaffeToGIEModle`?

since images in VOC benchmark are all not int the same resolution, I have to change the input dim before forwarding.

Is it possible to do this in TensorRT?