dusty-nv/jetson-inference #122: jetson-inference in TX2 segmentation failt(core dump) when call opencv imshow()

symao, 2017-08-03:
I try to use jetson-inference and opencv, it dumps once i call imshow(). But if i call other opencv functions like cv::Mat, it's ok.
I found it dumps before entering the main(), and i use gdb to debug, it says: 
> qmetaobject::className segmentation fault from libqt5core.so.5
I think jetson-inference use qt4, Does opencv imshow() use qt5? I assume it may be the conflict between qt4 and qt5, but how can i solve it?
Anybody run successfully with jetson-inference and opencv? Is there any sample code to do that?

alexlow-bh, 2019-02-22:
I am actually having this issue as well, there are issues from qt4 as well. The gdb debug gives a different error as below:

`QBasicAtomicInt::ref (this=<optimized out>)
    at ../../include/QtCore/../../src/corelib/arch/qatomic_aarch64.h:116
116	../../include/QtCore/../../src/corelib/arch/qatomic_aarch64.h: No such file or directory.`

The error is seen with opencv namedWindow function as well.

dusty-nv, 2019-02-22:
Can you try calling cudaThreadSynchronize() before you do OpenCV stuff?  Are you sure you are passing CPU pointer to OpenCV?

alexlow-bh, 2019-02-25:
Below is a code snippet of how the error is produced; the segmentation fault is thrown even before any code is run.

Thanks for taking the time to reply.

```
detectNet* net = detectNet::Create(argc, argv);
	
cv::VideoCapture cap(videosource);

if(!cap.isOpened())
{
	printf("Error opening video\n");
	return -1;
}

while(1)
{
	cv::Mat frame;
	cap >> frame;
	if (frame.empty())
		return 0;
	cv::imshow("Detections", frame);
	cv::waitKey(1);
}
```

closed 2023-04-27 by dusty-nv