dusty-nv/jetson-inference #123: Error during Loading Custom Models on Jetson

AndrewVGr, 2017-08-03:
I've got the following system:

Host:
====

Caffe:
--------
I0803 19:09:32.304337 22342 caffe.cpp:470] This is NVCaffe 0.16.3 started at Thu Aug  3 19:09:32 2017
I0803 19:09:32.304484 22342 caffe.cpp:473] CuDNN version: 6021
I0803 19:09:32.304488 22342 caffe.cpp:474] CuBLAS version: 8000
I0803 19:09:32.304491 22342 caffe.cpp:475] CUDA version: 8000
I0803 19:09:32.304492 22342 caffe.cpp:476] CUDA driver version: 8000

Digits:
--------
 ___ ___ ___ ___ _____ ___
 |   \_ _/ __|_ _|_   _/ __|
 | |) | | (_ || |  | | \__ \
 |___/___\___|___| |_| |___/ 6.0.0-rc.1

2017-08-03 19:11:14 [INFO ] Loaded 2 jobs.

Jetson TX2 with JetPack 3.1 (flashed to 28.1)

Training for 12 subclasses was successfully completed (I used NCCL). But after downloading model and trying to apply it to the jetson-inference on TX, I've got the following list of error messages:

`
imagenet-consolentu:~/jetson-inference/build/aarch64/bin$ ./imagenet-console bird_0.jpg output_0.jpg --prototxt=$NET/deploy.prototxt --model=$NET/snapshot_iter_178830.caffemodel --labels=$NET/labels.txt --input_b  args (8):  0 [./imagenet-console]  1 [bird_0.jpg]  2 [output_0.jpg]  3 [--prototxt=networks/GoogleNet-ILSVRC12-subset-II/deploy.prototxt]  4 [--model=networks/GoogleNet-ILSVRC12-subset-II/snapshot_iter_178830.caffemodel]  5 [--labels=networks/GoogleNet-ILSVRC12-subset-II/labels.txt]  6 [--input_blob=data]  7 [--output_blob=softmax]  

imageNet -- loading classification network model from:
         -- prototxt     networks/GoogleNet-ILSVRC12-subset-II/deploy.prototxt
         -- model        networks/GoogleNet-ILSVRC12-subset-II/snapshot_iter_178830.caffemodel
         -- class_labels networks/GoogleNet-ILSVRC12-subset-II/labels.txt
         -- input_blob   'data'
         -- output_blob  'softmax'
         -- batch_size   2

[GIE]  attempting to open cache file networks/GoogleNet-ILSVRC12-subset-II/snapshot_iter_178830.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading networks/GoogleNet-ILSVRC12-subset-II/deploy.prototxt networks/GoogleNet-ILSVRC12-subset-II/snapshot_iter_178830.caffemodel
[GIE]  retrieved output tensor 'softmax'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  conv1/7x7_s2: kernel weights has count 1 but 9408 was expected
[GIE]  conv2/3x3_reduce: kernel weights has count 1 but 4096 was expected
[GIE]  conv2/3x3: kernel weights has count 1 but 110592 was expected
[GIE]  inception_3a/1x1: kernel weights has count 1 but 12288 was expected
[GIE]  inception_3a/3x3_reduce: kernel weights has count 1 but 18432 was expected
[GIE]  inception_3a/3x3: kernel weights has count 1 but 110592 was expected
[GIE]  inception_3a/5x5_reduce: kernel weights has count 1 but 3072 was expected
[GIE]  inception_3a/5x5: kernel weights has count 1 but 12800 was expected
[GIE]  inception_3a/pool_proj: kernel weights has count 1 but 6144 was expected
[GIE]  inception_3b/1x1: kernel weights has count 1 but 32768 was expected
[GIE]  inception_3b/3x3_reduce: kernel weights has count 1 but 32768 was expected
[GIE]  inception_3b/3x3: kernel weights has count 1 but 221184 was expected
[GIE]  inception_3b/5x5_reduce: kernel weights has count 1 but 8192 was expected
[GIE]  inception_3b/5x5: kernel weights has count 1 but 76800 was expected
[GIE]  inception_3b/pool_proj: kernel weights has count 1 but 16384 was expected
[GIE]  inception_4a/1x1: kernel weights has count 1 but 92160 was expected
[GIE]  inception_4a/3x3_reduce: kernel weights has count 1 but 46080 was expected
[GIE]  inception_4a/3x3: kernel weights has count 1 but 179712 was expected
[GIE]  inception_4a/5x5_reduce: kernel weights has count 1 but 7680 was expected
[GIE]  inception_4a/5x5: kernel weights has count 1 but 19200 was expected
[GIE]  inception_4a/pool_proj: kernel weights has count 1 but 30720 was expected
[GIE]  inception_4b/1x1: kernel weights has count 1 but 81920 was expected
[GIE]  inception_4b/3x3_reduce: kernel weights has count 1 but 57344 was expected
[GIE]  inception_4b/3x3: kernel weights has count 1 but 225792 was expected
[GIE]  inception_4b/5x5_reduce: kernel weights has count 1 but 12288 was expected
[GIE]  inception_4b/5x5: kernel weights has count 1 but 38400 was expected
[GIE]  inception_4b/pool_proj: kernel weights has count 1 but 32768 was expected
[GIE]  inception_4c/1x1: kernel weights has count 1 but 65536 was expected
[GIE]  inception_4c/3x3_reduce: kernel weights has count 1 but 65536 was expected
[GIE]  inception_4c/3x3: kernel weights has count 1 but 294912 was expected
[GIE]  inception_4c/5x5_reduce: kernel weights has count 1 but 12288 was expected
[GIE]  inception_4c/5x5: kernel weights has count 1 but 38400 was expected
[GIE]  inception_4c/pool_proj: kernel weights has count 1 but 32768 was expected
[GIE]  inception_4d/1x1: kernel weights has count 1 but 57344 was expected
[GIE]  inception_4d/3x3_reduce: kernel weights has count 1 but 73728 was expected
[GIE]  inception_4d/3x3: kernel weights has count 1 but 373248 was expected
[GIE]  inception_4d/5x5_reduce: kernel weights has count 1 but 16384 was expected
[GIE]  inception_4d/5x5: kernel weights has count 1 but 51200 was expected
[GIE]  inception_4d/pool_proj: kernel weights has count 1 but 32768 was expected
[GIE]  inception_4e/1x1: kernel weights has count 1 but 135168 was expected
[GIE]  inception_4e/3x3_reduce: kernel weights has count 1 but 84480 was expected
[GIE]  inception_4e/3x3: kernel weights has count 1 but 460800 was expected
[GIE]  inception_4e/5x5_reduce: kernel weights has count 1 but 16896 was expected
[GIE]  inception_4e/5x5: kernel weights has count 1 but 102400 was expected
[GIE]  inception_4e/pool_proj: kernel weights has count 1 but 67584 was expected
[GIE]  inception_5a/1x1: kernel weights has count 1 but 212992 was expected
[GIE]  inception_5a/3x3_reduce: kernel weights has count 1 but 133120 was expected
[GIE]  inception_5a/3x3: kernel weights has count 1 but 460800 was expected
[GIE]  inception_5a/5x5_reduce: kernel weights has count 1 but 26624 was expected
[GIE]  inception_5a/5x5: kernel weights has count 1 but 102400 was expected
[GIE]  inception_5a/pool_proj: kernel weights has count 1 but 106496 was expected
[GIE]  inception_5b/1x1: kernel weights has count 1 but 319488 was expected
[GIE]  inception_5b/3x3_reduce: kernel weights has count 1 but 159744 was expected
[GIE]  inception_5b/3x3: kernel weights has count 1 but 663552 was expected
[GIE]  inception_5b/5x5_reduce: kernel weights has count 1 but 39936 was expected
[GIE]  inception_5b/5x5: kernel weights has count 1 but 153600 was expected
[GIE]  inception_5b/pool_proj: kernel weights has count 1 but 106496 was expected
[GIE]  loss3/classifier: kernel weights has count 1 but 12288 was expected
[GIE]  failed to build CUDA engine
failed to load networks/GoogleNet-ILSVRC12-subset-II/snapshot_iter_178830.caffemodel
failed to load networks/GoogleNet-ILSVRC12-subset-II/snapshot_iter_178830.caffemodel
imageNet -- failed to initialize.
`

AndrewVGr, 2017-08-09:
DIGITS 6.0 - the problem exists
DIGITS 5.0 - everything is allright

closed 2017-08-09 by AndrewVGr

mahxn0, 2017-12-12:
@AndrewVGr  did you solve this problem, i meet the same  question

mahxn0, 2017-12-12:
@AndrewVGr  
 did you solve this problem, i meet the same question
[image]

AndrewVGr, 2017-12-12:
When using the Digits 5.0 this issue is not present for me.