dusty-nv/jetson-inference #127: CMake stopped by error during downloading GoogleNet-ILSVRC12-subset.tar(and more)

silphy1001, 2017-08-08:
**[Environments]**
Device: Jetson TX1 with L4T R24.2.1(JetPack 2.3.1)
Host: Ubuntu 14.04 LTS x86_64
Commit for jetson-inference by Git: bed5c469866b43456ac7e910c8ed718e38ae7071

**[Issue]**
I refer to the Configuring with CMake section of https://github.com/dusty-nv/jetson-inference/blob/master/README.md#building-from-source-on-jetson.
When I execute "cmake ../", the following messages and waiting HTTP response.
But the Server never responds.
What should I try?

Show the following messages:
_Resolvinf public.boxcloud.com (public.boxcloud.com)... 107.152.25.200, 107.152.24.200
Connecting to public.boxcloud.com (public.boxcloud.com)|107.152.25.200|:443... conected.
WARNING: cannot verify public.boxcloud.com's certificate, issued by 'CN=GeoTrust SSL CA - G3,0=GeoTrust Inc.,C=US':
  Unable to locally verify the issuer's authority.
HTTP request sent, awaiting response... Read error (Connecting timed out) in headers.
Retrying._

Then, repeat retry, after these messages is shown,

**[Verify]**
I try to check following command line:
_wget --no-check-certificate 'https://nvidia.box.com/shared/static/zb8i3zcg39sdjjxfty7o5935hpbd64y4.gz' -O GoogleNet-ILSVRC12-subset.tar_

As a result, this command is failed by same error(awaiting response).

**[Others]**
In this report, I targeted **GoogleNet-ILSVRC12-subset**, but it seems that same error occured in other data such as **DetectNet-COCO-Airplane.tar.gz**.

dusty-nv, 2017-08-10:
Are you downloading from behind the Great Firewall of China by chance?   If so, you may need to manually download the models from Google Drive:   https://drive.google.com/drive/folders/0BwYxpotGWRNOV2d1NXNJOHA0dzQ?usp=sharing

For more info, see the previous issue https://github.com/dusty-nv/jetson-inference/issues/72

silphy1001, 2017-08-11:
Thank your for your reply, and sorry I could not find same issue such as #72.

To get straight to the point, I've succeeded downloading for all of files CMakePreBuild.sh needs.
(I comment out the lines described 'wget ...' in CMakePreBuild.sh, and I download all of files by using aria2c command and save the files in 'build' directory before executing 'cmake ../')
Thank your for your cooperation.

Finally, as reference, I describe my trial:
1.manual downloading from Google Drive
OK

2.manual downloading by aria2c command
OK

3.manual downloading by wget/curl command
NG:both wget/curl
Manual download by aria2c does work, but wget/curl does not work in same time.
If this issue is caused by a firewall, I think wget/curl/aria2c should fail.

4.behind the Great Firewall of China[GFW] by chance?
I think NO:
I work at Tokyo, Japan, and our office's internet has GLOBAL STATIC ipaddress.
If I fail **by chance**, I should succeed at least once, but I always fail.
Then I never think I download from behind the GFW **by chance**.

5.GFW blocks 'nvidia.box.com' and 'public.boxcloud.com'?
I think NO:
I tested 'nvidia.box.com' and 'public.boxcloud.com' using http://www.greatfirewallofchina.org/.
In result, both of urls are not blocked.

dusty-nv, 2017-08-11:
I just configured the repo again from scratch here (US), and it downloaded normally.  The reason it uses Box for the automated hosting is Box provides direct download link for use with wget.  Google Drive does not for large files...

longbowlee, 2017-08-13:
Hi dusty,
I think  GoogleNet-ILSVRC12-subset.tar is not uploaded in google drive

silphy1001, 2017-08-25:
Sorry for my late reply.

I don't say that URLs are wrong. In my environment, CMakePreBuild.sh is failed, but I have downloaded those files by "manually", too.
In my case, it seems that I can do another method without using wget(ex, using aria2-command, chromium-browser), so I have no problem in this case.
I will report you when I know the difference.

closed 2017-09-26 by dusty-nv