dusty-nv/jetson-inference #129: Using OpenCV Mat with jetson-inference

zainosaurus, 2017-08-09:
Hi,

I'm trying to use opencv to do inference on a trained model. I'm trying to pass in an image to the Detect method (in detectNet.cpp) and draw the bounding boxes on my Mat image. However, the Detect method takes in a float* as the image and I'm not sure how to format my Mat image so that it is a viable input. According to TensorRT documentation the image inputs should be in NCHW format (which is new to me), but when I try converting the images I consistently get errors like "cudaPreImageNetMean failed" or "tensorRT context failed". Could anyone provide some insight as to what I'm doing wrong, or if there's a better way to use OpenCV images for inference?

gariepyalex, 2017-08-22:
I managed to make it work. Let say cvImage is a 1920x1080 RGB uchar8 image. You first need to allocate memory on the GPU like this:
```c++
uchar3* imgBufferRGB = NULL;
float4* imgBufferRGBAf = NULL;
cudaMalloc((void**)&imgBufferRGB, 1920 * sizeof(uchar3) * 1080);
cudaMalloc((void**)&imgBufferRGBAf, 1920 * sizeof(float4) * 1080);
```        
 Then, you transfer the image to the GPU and convert it to RGBAf (because it is what detectnet takes as an input, it could be more efficient to modify detectNet.cpp and convert directly from uchar8 to TensorRT format).

 ```c++
  cudaMemcpy2D((void*)imgBufferRGB, 1920*sizeof(uchar3), (void*)cvImage.data, cvImage.step,
               1920*sizeof(uchar3), 1080, cudaMemcpyHostToDevice);
  cudaRGBToRGBAf(imgBufferRGB, imgBufferRGBAf, 1920, 1080); // defined in cudaRGB.h
  int numBoundingBoxes = maxBoxes;
  if (net->detect((float*)imgBufferRGBAf, 1920, 1080, bbCPU, &numBoundingBoxes, confCPU)) {
    printf("%i bounding boxes detected\n", numBoundingBoxes);
  }
```

I am pretty new du CUDA and the code may be inefficient but I hope it helps

curiouser001, 2017-09-25:
@gariepyalex Hi, 
Do you share your code entirely?

mahxn0, 2017-11-07:
@gariepyalex can you share your code for how to use opencv

gariepyalex, 2017-11-07:
Sorry I can't share the code

chenzhi1992, 2017-11-22:
Thank you， @gariepyalex . I did what you said, but the boundingbox is different from the result of the detect-console sample，and it doesn't make much difference. I read Mat to GPU memory.Can you help me ? Thank you!

mindmad, 2019-05-07:
hello, @gariepyalex  that was helpful. but I still need to convert back from float4 to Mat.
I tried to use the same way but with cudaRGVAfToRGB but it gives me errors .can you help me on it.

the error is this: 
`
[cuda]   cudaGetLastError()
[cuda]      invalid pitch argument (error 12) (hex 0x0C)
[cuda]      /home/nvidia/Desktop/mod/jetson-inference/utils/cuda/cudaRGB.cu:136
OpenCV Error: Assertion failed (size.width>0 && size.height>0) in imshow, file /home/nvidia/build_opencv/opencv/modules/highgui/src/window.cpp, line 331
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/nvidia/build_opencv/opencv/modules/highgui/src/window.cpp:331: error: (-215) size.width>0 && size.height>0 in function imshow

Aborted (core dumped)

`

closed 2023-04-27 by dusty-nv