dusty-nv/jetson-inference #131: Graphical user interface development for jetson-inference

curiouser001, 2017-08-10:
Hello there,
I created DetectNet as a QT project. I did the necessary library additions and then started. I get an error like this:

:-1: error: cannot find -lcudart
:-1: error: collect2: error: ld returned 1 exit status

Where could I have made a mistake. Thank you...

raaka1, 2017-08-11:
Please check
https://askubuntu.com/questions/510176/usr-bin-ld-cannot-find-lcudart

*Ravi Kiran.K*
Office:+6598821443
Timezone:   GMT + 8:00
http://www.hackerstrip.com/
https://hackerstrip.exposure.co/
https://twitter.com/Hackerstrip
http://hackerstrip.com/hackerstrip.asc
https://www.facebook.com/pages/Hackerstripcom/
<https://www.facebook.com/pages/Hackerstripcom/493515067421647>

curiouser001, 2017-08-11:
Thank you for your help. This error has disappeared. But other errors occurred. 
 ```
error: gstCamera.o: undefined reference to symbol 'g_error_free'
/lib/x86_64-linux-gnu/libglib-2.0.so.0:-1: error: error adding symbols: DSO missing from command line
error: collect2: error: ld returned 1 exit status
```

I tried to apply gst-camera to QT for simplicity. But I get the same mistake there. I think CMakeLists.txt files are a relevant issue.

dusty-nv, 2017-09-26:
`g_error_free()` is from GLib which is a dependency of GStreamer.  The `gstCamera.h` header uses some of those functions inline for error checking macros.  

Have you tried linking to `gstreamer-1.0` or similar like in [CMakeLists.txt:64](https://github.com/dusty-nv/jetson-inference/blob/f41f1c06f8fd31e17069d69f5822e0864026ac2f/CMakeLists.txt#L64)?

closed 2017-09-26 by dusty-nv