dusty-nv/jetson-inference #134: [GIE]  platform does not have FP16 support.

zmonoid, 2017-08-28:
I am running jetson-inference on Drive PX2(consists of 2 TX2), however it seems that this platform does not support FP16 (which should support), I checked the code and it comes with this:

`builder->platformHasFastFp16()` actually return false.

What may be the reason?

One more issue comes from loading my pretrained network, which will give such printout:

```
segNet -- loading segmentation network model from:
       -- prototxt:   ./networks/20170801-162610-cc33_epoch_2.0//deploy.prototxt
       -- model:      ./networks/20170801-162610-cc33_epoch_2.0//snapshot_iter_9044.caffemodel
       -- labels:     ./networks/20170801-162610-cc33_epoch_2.0//fpv-labels.txt
       -- colors:     ./networks/20170801-162610-cc33_epoch_2.0//fpv-deploy-colors.txt
       -- input_blob  'data'
       -- output_blob 'score_fr_21classes'
       -- batch_size  2

[GIE]  attempting to open cache file ./networks/20170801-162610-cc33_epoch_2.0//snapshot_iter_9044.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  enabled FP16 YES
[GIE]  platform has FP16 NO
[GIE]  platform does not have FP16 support.
[GIE]  loading ./networks/20170801-162610-cc33_epoch_2.0//deploy.prototxt ./networks/20170801-162610-cc33_epoch_2.0//snapshot_iter_9044.caffemodel
conv1: ERROR - 32-bit weights not found for 32-bit model
conv1: ERROR - 32-bit weights not found for 32-bit model
[GIE]  Parameter check failed at: Network.cpp::addConvolution::32, condition: kernelWeights.values != NULL
error parsing layer type Convolution index 1
[GIE]  failed to parse caffe network
failed to load ./networks/20170801-162610-cc33_epoch_2.0//snapshot_iter_9044.caffemodel
segNet -- failed to initialize.
segnet-console:   failed to initialize segnet
````

Could you give any suggestions?

Best,
Zhou Bin

dusty-nv, 2017-08-29:
Hi there,

platformHasFastFP16() should be returning true on PX2 iGPU (integrated GPU).   Can you confirm if you are using iGPU or dGPU (discrete GPU)?

Thank you,
Dustin

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

zmonoid, 2017-08-30:
Hi Dustin,

Yes, after using iGPU, it returns 0.

However, it seems iGPU is much slower than dGPU. In that case, using FP16/INT8 model to speed up the inference time will be less meaningful. What a pity.

Thank you,
Bin

zmonoid, 2017-08-30:
Hi Dustin,

About the second issue, it is similar to https://github.com/dusty-nv/jetson-inference/issues/84

I will try to use caffe-0.15 to solve it.

Best Regards,
Zhou Bin

dusty-nv, 2017-08-30:
Thank you Bin.  Since I am from NVIDIA Jetson team, I recommend contacting your DRIVE PX sales representative if you have further feedback.  My guess is future generations will align with FP16/INT8 acceleration throughout.

zmonoid, 2017-09-02:
Hi Dustin,

After using caffe-0.15, it will have problem when parsing Batch Normalization Layer of caffe, which gives following error:

'''
[GIE]  loading ./networks/ParkSeg/new_deploy.prototxt ./networks/ParkSeg/new_model.caffemodel
segnet-console: caffeParser.cpp:861: nvinfer1::ILayer* parseBatchNormalization(nvinfer1::INetworkDefinition&, const ditcaffe::LayerParameter&, CaffeWeightFactory&, BlobNameToTensor&): Assertion `mean.count == variance.count && movingAverage.count == 1' failed.
'''

It has been checked that 
```
net.params['bn1_1'][0].data 
net.params['bn1_1'][1].data
net.params['bn1_1'][2].data
net.params['bn1_1'][3].data
```
all have same length of 16.
The assertion error must come from movingAverage.count == 1.
However we don't have the source code of TensorRT to debug. Could you suggest what may be a solution?

Best,
Zhou Bin

closed 2023-04-27 by dusty-nv