dusty-nv/jetson-inference #135: Capture Detected İmage on DetectNet-Camera

curiouser001, 2017-08-29:
Hi, I want to capture some images from the detectNet-camera. I am using the following code from DetectNet-Console, but it did not produce a proper image:

`saveImageRGBA(argv[2], (float4*)imgCPU, imgWidth, imgHeight, 255.0f)` 

How can I do this? Thank you...

Enouk, 2017-09-12:
if( !camera->ConvertRGBA(imgCUDA, &imgRGBA, true) )
  printf("detectnet-camera:  failed to convert from NV12 to RGBA\n");

saveImageRGBA(argv[2], (float4*)imgRGBA, imgWidth, imgHeight, 255.0f)

Unless the third parameter is true, the memory will be allocated as CUDA device memory (i.e. not accessible as CPU)

barty777, 2017-09-15:
@Enouk Hi, 
I would like to do a center crop of the image but am stuck with the image format. I know that the image is contained in the `imgRGBA` variable but I have no clue how to get image data from it. Could you be kind to elaborate a bit how would I proceed with this problem?

dusty-nv, 2017-09-15:
For an example, take a look at this function:

https://github.com/dusty-nv/jetson-inference/blob/f41f1c06f8fd31e17069d69f5822e0864026ac2f/util/loadImage.cpp#L30

It reads imgRGBA and converts it to QT image.  Here’s the relevant part:

```for( int y=0; y < height; y++ )
{
for( int x=0; x < width; x++ )
               {
                              const float4 px = imgRGBA[y * width + x];
               }
}```

As in the function, you may need to re-scale the values to be 0.0-1.0 or 0-255.

From: Bartol Freškura [mailto:notifications@github.com]
Sent: Friday, September 15, 2017 9:57 AM
To: dusty-nv/jetson-inference <jetson-inference@noreply.github.com>
Cc: Subscribed <subscribed@noreply.github.com>
Subject: Re: [dusty-nv/jetson-inference] Capture Detected İmage on DetectNet-Camera (#135)

@Enouk<https://github.com/enouk> Hi,
I would like to do a center crop of the image but am stuck with the image format. I know that the image is contained in the imgRGBA variable but I have no clue how to get image data from it. Could you be kind to elaborate a bit how would I proceed with this problem?

—
You are receiving this because you are subscribed to this thread.
Reply to this email directly, view it on GitHub<https://github.com/dusty-nv/jetson-inference/issues/135#issuecomment-329790201>, or mute the thread<https://github.com/notifications/unsubscribe-auth/AOpDK_1qpEbN4ssEThLGEWyct_dvhiIBks5sioIlgaJpZM4PGEqz>.

-----------------------------------------------------------------------------------
This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

curiouser001, 2017-09-15:
@Enouk Hi, Thank you for your answer.
So, is there a way to save images of perceived objects?

barty777, 2017-09-25:
Hi @dusty-nv,

I'm trying to do a center crop of the captured camera frame before it goes through the detection module. I have tried your suggestion but keep getting the segmentation error. The code below is part of the `detectnet-camera.cpp` module. What am I doing wrong?

```
...
    while (!signal_recieved) {
        void *imgCPU = NULL;
        void *imgCUDA = NULL;

        // get the latest frame
        if (!camera->Capture(&imgCPU, &imgCUDA, 1000))
            printf("\ndetectnet-camera:  failed to capture frame\n");

        // convert from YUV to RGBA
        void *imgRGBA = NULL;

        if (!camera->ConvertRGBA(imgCUDA, &imgRGBA))
            printf("detectnet-camera:  failed to convert from NV12 to RGBA\n");

	for (int y=0; y < VIDEO_HEIGHT; y++)
	{
		for( int x=0; x < VIDEO_WIDTH; x++ )
		{
			const float4 px = ((float4*)imgRGBA)[y * VIDEO_WIDTH + x];
			printf("R: %f, G: %f, B: %f\n", px.x, px.y, px.z);
		}
	}
...
```

dusty-nv, 2017-09-25:
Hi, it looks like you need to pass 'true' as a third parameter to camera->ConvertRGBA ().  This will make the pointer accessible from CPU so it doesn't segfault.

barty777, 2017-09-25:
@dusty-nv 
I have created a cropped image array but when I send this array to the `DetectNet` function instead of the `imageRGBA`, I get the following error:
```
[GIE]  cudnnEngine.cpp (268) - Cuda Error in execute: 4
[GIE]  detectNet::Classify() -- failed to execute tensorRT context
[cuda]   cudaGetLastError()
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/Projects/jetson-inference/util/cuda/cudaRGB.cu:44
[cuda]   cudaRGBToRGBAf((uchar3*)input, (float4*)mRGBA[mLatestRGBA], mWidth, mHeight)
[cuda]      unspecified launch failure (error 4) (hex 0x04)
[cuda]      /home/ubuntu/Projects/jetson-inference/util/camera/gstCamera.cpp:114
detectnet-camera:  failed to convert from NV12 to RGBA
[1]    3423 segmentation fault  ./detectnet-camera --prototxt=$NET/deploy.prototxt  --input_blob=data 
```
The current code goes something like this:
```
...
void* new_array = NULL;
void* gpuPtr = NULL;
const size_t size = new_width * new_height * sizeof(float4);

if( !cudaAllocMapped(&new_array, &gpuPtr, size) )
{
	printf(LOG_CUDA "gstCamera -- failed to allocate zeroCopy memory for %ux%xu RGBA texture\n", new_width, new_height);
	return false;
}
if( new_array != gpuPtr )
{
	printf(LOG_CUDA "gstCamera -- zeroCopy memory has different pointers, please use a UVA-compatible GPU\n");
	return false;
}

...
// get the latest frame
if (!camera->Capture(&imgCPU, &imgCUDA, 1000))
            printf("\ndetectnet-camera:  failed to capture frame\n");

 // convert from YUV to RGBA
void *imgRGBA = NULL;

if (!camera->ConvertRGBA(imgCUDA, &imgRGBA, true))
  printf("detectnet-camera:  failed to convert from NV12 to RGBA\n");

for (int y=0; y < VIDEO_HEIGHT; y++)
{
	for( int x=0; x < VIDEO_WIDTH; x++ )
	{
		// Fill the new_array with desired values code goes here
	}
}

// classify image with detectNet
int numBoundingBoxes = maxBoxes;

if( net->Detect((float*)new_array, camera->GetWidth(), camera->GetHeight(), bbCPU, &numBoundingBoxes, confCPU))
...

```

Is there maybe a simpler way to do the center crop?

dusty-nv, 2017-09-25:
I think the problem may be that you are passing original camera width and height to net->Detect() and not the new cropped width and height.

After this CPU method works you would probably want to do the cropping in CUDA, either in its own kernel, or you could add it to the function in imageNet.cu which DetectNet uses.

zip520123, 2017-11-06:
I want to save image when facenet detect.
I insert code in  "if( nc != lastClass || n == (numBoundingBoxes - 1) )" scope.
`char filename[128];
  time_t now;
  struct tm tm_now;

  now = time(NULL); 
  localtime_r(&now, &tm_now);

   strftime(filename, sizeof(filename), "%Y%m%d_%H%M%s.jpg", &tm_now);
   int imgW = bb[2] - bb[0];
   int imgH = bb[3] - bb[1];
   saveImageRGBA(filename, (float4*)imgCPU, imgW , imgH,255.0f );
`
but saved image in desk only has noise pixel.
How can I fix it?