dusty-nv/jetson-inference #136: problem linking libjetson-inference.so

phojjat, 2017-09-06:
Hello,

I have built your examples on my jetson tx2 and everything works correctly. Now I am trying to build a standalone inference code based on the provided jet-inference code in Code blocks IDE. Receiving images from camera works okay, however when I extend it to include inference using libjetson-inference.so I cannot run the executable due to the following error "cannot open shared object file: No such file or directory". I have ran an ldd on the library and linked all the dependencies that show up, but that has not solved the issue. I am wondering what other steps I need to take to fix this. Any help is appreciated.

Thank you!

dusty-nv, 2017-09-06:
Hmm...are you using cmake to build your custom application, or else any differences in the link line?

You could try running `make install` or `sudo make install` after you run `make` on jetson-inference.  It will install the libs to the standard locations.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

phojjat, 2017-09-06:
I am using code blocks IDE which compiles and runs from within the application. I have linked libjetson-inference. <http://libjetson-inference.io/>so along with the dependencies I found through ldd but the issue persists.

dusty-nv, 2017-09-26:
As per [CMakeLists.txt:6](https://github.com/dusty-nv/jetson-inference/blob/f41f1c06f8fd31e17069d69f5822e0864026ac2f/CMakeLists.txt#L6), `jetson-inference` compiles with `-std=c++11` flag.  This is because TensorRT also uses c++11.  Are you building with that flag too from Code Blocks?

phojjat, 2017-09-26:
Yes. I solved the issue by adding the path of the library to the config file as per: https://lonesysadmin.net/2013/02/22/error-while-loading-shared-libraries-cannot-open-shared-object-file/ <https://lonesysadmin.net/2013/02/22/error-while-loading-shared-libraries-cannot-open-shared-object-file/>

danhnguyen12091998, 2020-05-23:
Hi @phojjat , I have the same issue with you, so can you show me the detailed instructions for fixing this issue ? Thank you so much

closed 2023-04-27 by dusty-nv