dusty-nv/jetson-inference #137: How to check FPS ..

vusdo, 2017-09-07:
If you look at the specification of the onboard camera, it says 120 fps for 1280 * 720.
Actual execution is expressed as follows.

NvCameraSrc: Trying To Set Default Camera Resolution. Selected sensorModeIndex = 2 WxH = 1280x720 FrameRate = 120.000000 ....

The display shows 60 FPS as follows.

TensorRT build 836 | FP16 | 60.0 FPS

Why is 120Frame not coming out?

The above code is FPS when you capture and run ConvertRGBA only.

dusty-nv, 2017-09-07:
Either you are only getting 60fps from gstreamer, or there is VSYNC with OpenGL and the display.  How are you setting the camera to 120fps?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

vusdo, 2017-09-09:
nvcamerasrc ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720,format=(string)I420, framerate=(fraction)120/1

dusty-nv, 2017-09-26:
Using the [gst-camera](https://github.com/dusty-nv/jetson-inference/blob/master/util/camera/gst-camera/gst-camera.cpp) test app or similar, please disable openGL display.

You can do it by making the following change to [gst-camera.cpp:81](https://github.com/dusty-nv/jetson-inference/blob/f41f1c06f8fd31e17069d69f5822e0864026ac2f/util/camera/gst-camera/gst-camera.cpp#L81)

```
--- glDisplay* display = glDisplay::Create();
+++ glDisplay* display = NULL;
```
This will make is so display isn't used to limit camera capture with VSYNC.
Closing this issue as FPS is displayed in the GUI title bar.

closed 2017-09-26 by dusty-nv