dusty-nv/jetson-inference #139: DrawBoxes (gpuRectOutlines) is a major bottleneck in performance

NicholasSieb, 2017-09-13:
Profiling both with Ubuntu desktop environment and Jetson TX2, the major bottlenecks seem to be gpuRectOutlines and RGBtoRGBAf. I managed to speed up the color conversion, but haven't had any luck in speeding up gpuRectOutlines. 

Drawing boxes is taking an average of 30+ms per call

58.09%  1.08518s        34  31.917ms  30.281ms  34.945ms  void gpuRectOutlines<float4>(float4*, float4*, int, int, float4*, int, float4)

Any ideas in clearing up this bottleneck?

dusty-nv, 2017-09-26:
For applications that only need to access the bounding box coordinates and not the overlay, have you tried disabling the visualization?

closed 2017-10-03 by NicholasSieb

kanakiyab, 2017-11-17:
Hi @NicholasSieb ,

I very much agree with you on your observation. I have disabled the visualizations as @dusty-nv suggested. 

Can you please share some insights on how did you manage to speed up the color conversion?

Thanks,
Bhargav