dusty-nv/jetson-inference #144: Video Classification / Activity-Recognition

dmorin1, 2017-10-06:
I've been experimenting with this project on Jetson TX1 for some time and found it's a very good base start. I'm interested by doing video classification. I modified imagenet-console to receive to video ( from file or rtsp stream) do classify each frame using googlenet. This is working, but I would like to go further and been able to classify video (spatio-temporal data) has opposed to static image. Did someone tried to video classification using TensorRT? Is there example of Temporal-Segment-Network (TSN) model used on TensorRT.  I'm hoping to do classification on UCF101 dataset.

I found a UCF101 pre-trained networks ( image+optical flow) on this project:
https://github.com/yjxiong/temporal-segment-networks

Unfortunately TensorRT does not support Batch Normalization layer used in this network.

I found a note in TensorRT documentation:
"Batch Normalization can be implemented using the TensorRT Scale layer."
Read more at: http://docs.nvidia.com/deeplearning/sdk/tensorrt-user-guide/index.html#ixzz4ul8v2yAv 
Follow us: @GPUComputing on Twitter | NVIDIA on Facebook

I'm note sure how to do the conversion from Batch Normalization to Scale layer. Any advice would be welcome on this subject.

Assuming I can find a way to convert Batch Normalization layer to Scale layer. Can I simply modify the prototxt file and keep using pre-trained network (caffemodel) without re-training?

Thank you.

pratesh, 2018-01-04:
Hi @dmorin1 , can you please share your modified imagenet-console that can receive video from a locally stored video file

LSAMIJN, 2019-10-22:
Hi dmorin1,

I'm trying for some time now to modify the imagenet-console to work with a rtsp stream (from ip camera) but cannot let it work so far?
is it possible to share me your code or to give me some directions how to let it work?
thank you