dusty-nv/jetson-inference #146: Make problem

Patrick0920, 2017-10-09:
/usr/bin/ld: warning: libcudnn.so.7, needed by /usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so, not found (try using -rpath or -rpath-link)
/usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so: undefined reference to `cudnnSetLRNDescriptor'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so: undefined reference to `cudnnGetConvolutionBackwardDataWorkspaceSize'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so: undefined reference to `cudnnSetFilter4dDescriptor'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so: undefined reference to `cudnnGetRNNLinLayerMatrixParams'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so: undefined reference to `cudnnCreateConvolutionDescriptor'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so: undefined reference to `cudnnDestroyConvolutionDescriptor'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so: undefined reference to `cudnnSetTensorNdDescriptor'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../../lib/libnvinfer.so: undefined reference to `cudnnSetFilterNdDescriptor'

XhqGlorry11, 2017-10-10:
@dusty-nv same problem here. I have installed cudnn in /usr/local/cuda-8.0. Any suggestions about what's going wrong?

deksprime, 2017-10-29:
Something's really gone wrong with libnvinfer nowadays. I've been experiencing another problem related to libnvinfer and have stumbled upon this issue while I was searching for a solution in google. The problem I'm having is that I'm not able to install tensorRT2.1.2 since it looks for this libnvinfer3 and I really don't know how to install it.

dusty-nv, 2017-10-30:
Is your Jetson TX1/TX2 running JetPack 3.1?   It should automatically install cuDNN and TensorRT 2.1 for you.

deksprime, 2017-10-30:
my bad ... I was trying to install from the website with the deb package

RoboEvangelist, 2018-04-28:
still. I used Jetpack 3.2, and I have the same probelm

closed 2023-04-27 by dusty-nv