dusty-nv/jetson-inference #147: make error

XhqGlorry11, 2017-10-10:
I run into an error when compiling the project on TX2. The error warnings are as follows. I have installed cudnn on TX2. I mean, cudnn.h is in /usr/local/cuda-8.0/include and libcudnn.so libcudnn.so.7 libcudnn.so.7.0.3 are in /usr/local/cuda-8.0/lib64. And /usr/local/cuda-8.0/lib64 has been added to LD_LIBRARY_PATH. I dont know what's going wrong. Any help is appreciated. THX!
[ 58%] Linking CXX executable ../aarch64/bin/imagenet-console
/usr/bin/ld: warning: libcudnn.so.7, needed by //usr/local/lib/libnvinfer.so, not found (try using -rpath or -rpath-link)
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetLRNDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnGetConvolutionBackwardDataWorkspaceSize'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetFilter4dDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnGetRNNLinLayerMatrixParams'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreateConvolutionDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyConvolutionDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetTensorNdDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetFilterNdDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnConvolutionForward'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyOpTensorDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnGetRNNLinLayerBiasParams'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetReduceTensorDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnGetReductionWorkspaceSize'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreateTensorDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyDropoutDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnRNNForwardInference'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnConvolutionBackwardData'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreateDropoutDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreatePoolingDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyFilterDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetOpTensorDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyTensorDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnGetRNNParamsSize'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreateLRNDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyRNNDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnGetConvolutionForwardWorkspaceSize'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnAddTensor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyPoolingDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnGetFilterNdDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSoftmaxForward'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnActivationForward'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnGetRNNWorkspaceSize'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnPoolingForward'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyReduceTensorDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnReduceTensor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetTensor4dDescriptorEx'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreateRNNDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDropoutGetStatesSize'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetPooling2dDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreateFilterDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetActivationDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetStream'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnLRNCrossChannelForward'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyLRNDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreateActivationDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetTensor4dDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroy'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnDestroyActivationDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetRNNDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetConvolution2dDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreateOpTensorDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnSetDropoutDescriptor'
//usr/local/lib/libnvinfer.so: undefined reference to `cudnnCreate'
collect2: error: ld returned 1 exit status
imagenet-console/CMakeFiles/imagenet-console.dir/build.make:101: recipe for target 'aarch64/bin/imagenet-console' failed
make[2]: *** [aarch64/bin/imagenet-console] Error 1
CMakeFiles/Makefile2:122: recipe for target 'imagenet-console/CMakeFiles/imagenet-console.dir/all' failed
make[1]: *** [imagenet-console/CMakeFiles/imagenet-console.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

XhqGlorry11, 2017-10-10:
@dusty-nv please help~

dusty-nv, 2017-10-10:
I think you would have to add /usr/local/cuda/lib64 to the linker path in CMakeLists.txt

Per the tutorial, the project is setup for a JetPack install, which installs it under /usr/lib

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

XhqGlorry11, 2017-10-11:
@dusty-nv Hi, is cudnn for TX2 the same with cudnn for titanx? And when will nvidia developer zone(https://developer.nvidia.com/embedded/jetpack) finish maintenance? No download is available right now..... thx~

dusty-nv, 2017-10-11:
Yes the cuDNN API is the same, however it’s install location can vary by architecture.

The JetPack page is back up, let me know if you still have problems after clearing your browser cache.  Thanks.

XhqGlorry11, 2017-10-12:
@dusty-nv Hi, I know that TensorRT3 is equipped with python interface and where can we get python API documentation?

dusty-nv, 2017-10-12:
Hi, please download the x86_64 tarball from the RC page and look in the extracted python/doc folder for the HTML documentation.

XhqGlorry11, 2017-10-13:
@dusty-nv Hi dusty, I find the python API doc of tensorrt3 under your instruction and thx for that. It seems that python API doc of uff is not included in API doc of tensorrt3. Do you know where we can find the API doc of uff?

dusty-nv, 2017-10-13:
The UFF python parser documentation is included in the Package Reference at this page:

file:///C:/Users/dustinf/Downloads/tensorRT/TensorRT_3.0_RC/TensorRT-3.0.0.Ubuntu-16.04.3.cuda-9.0.x86_64/TensorRT-3.0.0/python/doc/python2.7/pkg_ref/parsers.html

That was a link to the docs on my computer, replace the path prefix with your own.  Or just expand ‘tensorrt.parsers’ on the left side of the HTML documentation under ‘Package Reference’.

I’m not sure where the file format documentation for UFF itself is, it came out of the TensorFlow project.  A quick google search didn’t immediately turn it up.

XhqGlorry11, 2017-10-13:
@dusty-nv Hi dusty, I find the python API doc of tensorrt3 under your instruction and thx for that. But it seems that current API doc is not completed. For example, in the demo given there are following codes:parser = uffparser.create_uff_parser(),parser.register_input("Placeholder", (1,28,28)),parser.register_output("fc2/Relu"). I can just search function "create_uff_parser()" in that doc. But I cannot find register_input() or register_output() because they are not included in the current API doc, which means that I don't know what methods and variables are provided with parser and of course don't know how to use it. Do you have any plan of completing the API doc to the full version? thx

dusty-nv, 2017-10-13:
Yes, it appears the register_input() and register_output() functions are missing from the release candidate documentation, however the provided example using them is now working, so hopefully you are able to proceed.  Understand that the package is effectively in beta until the general access (GA) release with the full documentation of course.

XhqGlorry11, 2017-10-14:
@dusty-nv Hi dusty, I need to use add_element_wise() to manually build a model. But this function just accept tensorrt.infer.tensor as its inputs, while the parameters I have for element_wise operation is numpy.nparray. So can you tell me how to convert a numpy.nparray format variable into tensorrt.infer.tensor format variable? And also the method backwards? thx