dusty-nv/jetson-inference #150: make error, ‘NV_TENSORRT_MAJOR’ was not declared in this scope

lengly, 2017-10-19:
Hi @dusty-nv 
I finish "Running JetPack on the Host" step, and then I try to build the source on Jetson TX2.
But I met such error when make:
Is it because something wrong with my tensorrt?

```
/home/nvidia/git/jetson-inference/tensorNet.cpp: In member function ‘bool tensorNet::LoadNetwork(const char*, const char*, const char*, const char*, const std::vector<std::__cxx11::basic_string<char> >&, uint32_t)’:
/home/nvidia/git/jetson-inference/tensorNet.cpp:220:55: error: ‘NV_TENSORRT_MAJOR’ was not declared in this scope
  printf(LOG_GIE "TensorRT version %u.%u, build %u\n", NV_TENSORRT_MAJOR, NV_TEN
                                                       ^
/home/nvidia/git/jetson-inference/tensorNet.cpp:220:74: error: ‘NV_TENSORRT_MINOR’ was not declared in this scope
 tf(LOG_GIE "TensorRT version %u.%u, build %u\n", NV_TENSORRT_MAJOR, NV_TENSORRT
                                                                     ^
CMakeFiles/jetson-inference.dir/build.make:1373: recipe for target 'CMakeFiles/jetson-inference.dir/tensorNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/tensorNet.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

lengly, 2017-10-20:
Solve the issue.
This repo use NV_TENSORRT_MAJOR variable to detect the version of TensorRT, (version 1, 2, 3). But in TensorRT version 1, there is no such variable...

dusty-nv, 2017-10-30:
Thanks for reporting this, I have fixed the issue with TensorRT version 1 in commit [f0f3c35](https://github.com/dusty-nv/jetson-inference/commit/f0f3c35c94997dea3072b1e4c8f406057d778058).

Let me know if you experience further issue.

closed 2017-10-30 by dusty-nv