dusty-nv/jetson-inference #154: Make problem C++ requires whitespace after the macro name

BazzzNL, 2017-10-31:
Hi dusty and others, I just installed this following the provided guide on my Jetson TX2 using Ubuntu 16.06 LTS but at the point where one should use `make` I get the following problems. Any idea what could be wrong?

nvidia@tegra-ubuntu:~/jetson-inference/build$ make
[  2%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaYUV-NV12.cu.o
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[  5%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/jetson-inference_generated_imageNet.cu.o
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[  7%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaNormalize.cu.o
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[ 10%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaRGB.cu.o
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[ 12%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaResize.cu.o
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:7: warning: ISO C++11 requires whitespace after the macro name
[ 15%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/util/cuda/jetson-inference_generated_cudaOverlay.cu.o
<command-line>:0:2: warning: ISO C++11 requires whitespace after the macro name
[... 85 lines cut ...]
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmUnmap'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevices'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmGetDevice'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmOpenOnce'
/usr/lib/gcc/aarch64-linux-gnu/5/../../../aarch64-linux-gnu/libGL.so: undefined reference to `drmFreeDevices'
collect2: error: ld returned 1 exit status
imagenet-console/CMakeFiles/imagenet-console.dir/build.make:101: recipe for target 'aarch64/bin/imagenet-console' failed
make[2]: *** [aarch64/bin/imagenet-console] Error 1
CMakeFiles/Makefile2:122: recipe for target 'imagenet-console/CMakeFiles/imagenet-console.dir/all' failed
make[1]: *** [imagenet-console/CMakeFiles/imagenet-console.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
nvidia@tegra-ubuntu:~/jetson-inference/build$ 

g++ -v

nvidia@tegra-ubuntu:~/jetson-inference/build$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-linux-gnu/5/lto-wrapper
Target: aarch64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.5' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-arm64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-arm64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-arm64 --with-arch-directory=aarch64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-multiarch --enable-fix-cortex-a53-843419 --disable-werror --enable-checking=release --build=aarch64-linux-gnu --host=aarch64-linux-gnu --target=aarch64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.5)

dusty-nv, 2017-10-31:
Hi BazzzNL, was your TX2 flashed with a recent JetPack (version 3.1 is the latest) and did you allow it to install cuDNN and TensorRT?  Thanks.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

BazzzNL, 2017-10-31:
Hi Dusty, yes I just flashed it today, indeed with JetPack v3.1. I followed your guide in readme.md on the front page and also the one that you link to at [docs.nvidia.com](http://docs.nvidia.com/jetpack-l4t/index.html#developertools/mobile/jetpack/l4t/3.0/jetpack_l4t_install.htm). I selected JetPack 3.1, TX2 and Full Installation. It went through a rather long proces of flashing the device and then sending over packages via ssh. I do believe cuDNN and TensorRT are amongst these packages as I tried the Cuda OceanFFT sample on the TX2 and that worked fine. Therefore, I continued with your guide and started the section named 'Building from Source on Jetson' and got the before mentioned issues.
Perhaps the funny thing is that imagenet-camera seems to work using the googlenet default set, but any command line parameters are not parsed.

dusty-nv, 2017-10-31:
Hmm I just rebuilt yesterday on TX2/JetPack 3.1 but did not see the issue.  It sounds like you are experiencing an issue similar to:

[https://devtalk.nvidia.com/default/topic/946136/building-an-opengl-application/](https://devtalk.nvidia.com/default/topic/946136/building-an-opengl-application/)

where mesa libGL is potentially interfering with NVIDIA libGL.   You may also want to try adding `drm` to `target_link_libraries` in CMakeLists.txt here:   [https://github.com/dusty-nv/jetson-inference/blob/f0f3c35c94997dea3072b1e4c8f406057d778058/CMakeLists.txt#L64](https://github.com/dusty-nv/jetson-inference/blob/f0f3c35c94997dea3072b1e4c8f406057d778058/CMakeLists.txt#L64)

Sorry, I am OOTO/AFK for the next couple days so you may have to try some things on your board.

closed 2023-04-27 by dusty-nv