dusty-nv/jetson-inference #156: Building from Source on the Jetson - CMAKE

stormchaserjohn, 2017-11-04:
Have followed the directions on the repo for building from source on the Jetson, and I after I create the build directory and attempt to run "cmake../", I keep getting the error "bash: cmake../: No such file or directory. 

I have tried downloading the repo several times, checking ~/.bashrc and so on, and just can't find a fix for this.

elabeca, 2017-11-04:
You’ll need to put a space between “cmake” and the parent folder “../“

stormchaserjohn, 2017-11-04:
OMG, (hangs head in shame) can't believe it was that simple. Thanks so much for pointing it out, you made my day, sir!

closed 2017-11-04 by stormchaserjohn

abdallah0012, 2019-04-26:
hello
when i run cmake ../ ; i have this error.

CMake Error at /usr/share/cmake-3.10/Modules/FindQt4.cmake:1320 (message):
  Found unsuitable Qt version "5.9.5" from /usr/bin/qmake, this code requires
  Qt 4.x
Call Stack (most recent call first):
  utils/CMakeLists.txt:12 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/jetson-inference/build/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/jetson-inference/build/CMakeFiles/CMakeError.log".

Please, help me.

dusty-nv, 2019-04-26:
Hi @abdallah0012 , please try running these commands before trying again:

```
sudo apt-get update
sudo apt-get install -y libqt4-dev qt4-dev-tools libglew-dev glew-utils libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev
sudo apt-get install -y libopencv-calib3d-dev libopencv-dev
```

abdallah0012, 2019-04-27:
Thanks Dustin for your reply

It resolve my problem but when i run ./detectnet-camera with my nvidia agx
xavier card and logitech c930e i have this error.
detectnet-camera: failed to capture frame
detectnet  -camera: failed to convert from NV12 to RGBA
detectNet::Classify( 0x(nil), 1280, 720 ) -> invalid parameters
[cuda] cudaNormalizeRGBA((float4*)imgRGBA, make_float2(0.0f, 255.0f),
(float4*)imgRGBA, make_float2(0.0f, 1.0f), camera->GetWidth(),
camera->GetHeight())
[cuda] invalid device pointer (error 17) (hex 0x11)
[cuda] /home/nvidia/jetson-inference/detectnet-camera/detectnet
-camera.cpp:247

Can you help me please !!

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>