dusty-nv/jetson-inference #157: how to use opencv convert imgCPU to mat

mahxn0, 2017-11-07:
`#include <unistd.h>
#include "loadImage.h"
#include "cudaNormalize.h"
#include "opencv2/opencv.hpp"
#include "opencv2/highgui/highgui.hpp"
using namespace cv;
bool signal_recieved = false;
Mat nv12tomat(unsigned char*yuvdata,unsigned int width,unsigned int height)
{
	Mat nv12_img=Mat(height*3/2,width,CV_8UC1,yuvdata);
	Mat rgb_img;
	cvtColor(nv12_img,rgb_img,CV_YUV2BGR_NV12);
	return rgb_img;
}
void sig_handler(int signo)
{
	if( signo == SIGINT )
	{
		printf("received SIGINT\n");
		signal_recieved = true;
	}
}

int main( int argc, char** argv )
{
	printf("gst-camera\n  args (%i):  ", argc);

	for( int i=0; i < argc; i++ )
		printf("%i [%s]  ", i, argv[i]);
		
	printf("\n");
	
		
	if( signal(SIGINT, sig_handler) == SIG_ERR )
		printf("\ncan't catch SIGINT\n");

	/*
	 * create the camera device
	 */
	gstCamera* camera = gstCamera::Create();
	
	if( !camera )
	{
		printf("\ngst-camera:  failed to initialize video device\n");
		return 0;
	}
	u_int32_t m_Width=camera->GetWidth();
	u_int32_t m_Height=camera->GetHeight();
	u_int32_t m_Bitdepth=camera->GetPixelDepth();
	printf("\ngst-camera:  successfully initialized video device\n");
	printf("   width:  %u\n", m_Width);
	printf("   height:  %u\n", m_Height);
	printf("   depth:  %u (bpp)\n", m_Bitdepth);
	if( !camera->Open() )
	{
		printf("\ngst-camera:  failed to open camera for streaming\n");
		return 0;
	}
	printf("\ngst-camera:  camera open for streaming\n");
	//FILE *fp=fopen("1.yuv","w+");
	void *imgCUDA=NULL;
	void *imgCPU=NULL;
	while(!signal_recieved)
	{
		struct timeval tvs,tve;
		clock_t start,end;
		start=clock();
		gettimeofday(&tvs,NULL);
		// get the latest frame
		if(!camera->Capture(&imgCPU,&imgCUDA,1000))
		{
			printf("\ngst-camera: failed to capture frame\n");
		}
		else
			printf("gst-camera:  recieved new frame  CPU=0x%p  GPU=0x%p\n",imgCPU,imgCUDA);
			//fwrite(imgCPU,m_Width*m_Height*3/2,1,fp);
		// if(!camera->ConvertRGBA(imgCUDA,&imgRGBA,1))
		// {
		// 	printf("gst-camera:  failed to convert from NV12 to RGBA\n");
		// }		  
		//Mat img(1280,720,CV_8U,imgCPU);
		
	//	nv12tomat((unsigned char *)imgCPU,m_Width,m_Height);
		gettimeofday(&tve,NULL);
	    double span = tve.tv_sec-tvs.tv_sec + (tve.tv_usec-tvs.tv_usec)/1000000.0;
    	printf("gettimeofday time: %.12f\n",span);
		end=clock();
		std::cout<<"end-start="<<end-start<<std::endl;
		//printf("img height=%d\n",img.cols);
		//printf("img width=%d\n",img.rows);
		//imwrite("1.bmp",img);`

this is my code ,but i found  this take 80ms for convert,  can you give me some advice    thank you

hhamii2040, 2018-06-04:
hi
for imgcpu this code is OK
	cv::Mat gray_img = Mat(1080,1920,CV_8UC1 ,(unsigned char*)imgCPU);
	if(!gray_img .empty())
    	{
		cv::imwrite("/home/nvidia/Desktop/Pre_Proc.png", gray_img);
	}//if

mahxn0, 2018-06-06:
@hhamii2040  thank you

closed 2018-06-21 by mahxn0