dusty-nv/jetson-inference #163: DetectNet Camera Error!

curiouser001, 2017-11-21:
Hello,
I am trying to run DetectNet-Camera on my main machine. I've worked it before. It did not work after reinstalling the computer. I'm getting the following errors.

`./detectnet-camera 
detectnet-camera
  args (1):  0 [./detectnet-camera]  

[gstreamer] initialized gstreamer, version 1.2.4.0
[gstreamer] gstreamer decoder pipeline string:
v4l2src device=/dev/video0 ! video/x-raw, width=(int)1280, height=(int)720, format=RGB ! videoconvert ! video/x-raw, format=RGB ! videoconvert !appsink name=mysink

detectnet-camera:  successfully initialized video device
    width:  1280
   height:  720
    depth:  24 (bpp)

detectNet -- loading detection network model from:
          -- prototxt    /home/monster/Desktop/jetson-inference/build/x86_64/bin/networks/final/deploy.prototxt
          -- model       /home/monster/Desktop/jetson-inference/build/x86_64/bin/networks/final/snapshot_iter_27552.caffemodel
          -- input_blob  'data'
          -- output_cvg  'coverage'
          -- output_bbox 'bboxes'
          -- threshold   0.500000
          -- batch_size  2

[GIE]  TensorRT version 2.1, build 2102
[GIE]  attempting to open cache file /home/monster/Desktop/jetson-inference/build/x86_64/bin/networks/final/snapshot_iter_27552.caffemodel.2.tensorcache
[GIE]  loading network profile from cache... /home/monster/Desktop/jetson-inference/build/x86_64/bin/networks/final/snapshot_iter_27552.caffemodel.2.tensorcache
[GIE]  platform does not have FP16 support.
[GIE]  /home/monster/Desktop/jetson-inference/build/x86_64/bin/networks/final/snapshot_iter_27552.caffemodel loaded
detectnet-camera: cudnnEngine.cpp:605: bool nvinfer1::cudnn::Engine::deserialize(const void*, std::size_t, nvinfer1::IPluginFactory*): Assertion `size >= bsize && "Mismatch between allocated memory size and expected size of serialized engine."' failed.
Aborted (core dumped)
`

ashishmokalkar, 2017-12-05:
@dusty-nv Has anyone been able to solve this issue? I am using custom car detection model on jetpack 3.1 on TX2. On jetpack 3.0, I didnt have this issue. It was running perfectly. But after updating, it threw this error.

jtrovato, 2017-12-06:
@curiouser001 did you update anything when "reinstalling the computer"? I am experiencing the same issue after upgrading Tensor RT and CuDnn.

closed 2023-04-27 by dusty-nv