dusty-nv/jetson-inference #169: DetectNet model: invalid device function on NVIDIA DIGITS 6 AMI

alamesa, 2017-12-01:
Hello, 

**Overview**
I am working through the "Locating Object Coordinates using DetectNet" section of the tutorial.  I am using the DIGITS 6 AMI on Ubuntu 16.04 from the AWS Marketplace (instance type is gx.2xlarge), but ran into a roadblock when trying to launch the DetectNet-COCO-Dog model. At a high level, the error that I'm seeing is: "ERROR: Check failed: error == cudaSuccess (8 vs. 0) invalid device function." 

**More details:**
Here are the first and last several lines of output from caffe_output.log for the job: 

[snip]
libdc1394 error: Failed to initialize libdc1394
I1126 11:55:07.177713 579 upgrade_proto.cpp:1044] Attempting to upgrade input file specified using deprecated 'solver_type' field (enum)': /jobs/20171126-115505-4756/solver.prototxt
I1126 11:55:07.177882 579 upgrade_proto.cpp:1051] Successfully upgraded file specified using deprecated 'solver_type' field (enum) to 'type' field (string).
W1126 11:55:07.177889 579 upgrade_proto.cpp:1053] Note that future Caffe releases will only support 'type' field (string) for a solver's type.
I1126 11:55:07.234048 579 caffe.cpp:197] Using GPUs 0
I1126 11:55:07.234225 579 caffe.cpp:202] GPU 0: GRID K520
I1126 11:55:07.730933 579 solver.cpp:48] Initializing solver from parameters:
[/snip]

[snip]
F1126 11:55:09.247619 579 pooling_layer.cu:212] Check failed: error == cudaSuccess (8 vs. 0) invalid device function
*** Check failure stack trace: ***
@ 0x7f205acaddaa (unknown)
@ 0x7f205acadce4 (unknown)
@ 0x7f205acad6e6 (unknown)
@ 0x7f205acb0687 (unknown)
@ 0x7f205b4166a0 caffe::PoolingLayer<>::Forward_gpu()
@ 0x7f205b223358 caffe::Net<>::ForwardFromTo()
@ 0x7f205b2236d7 caffe::Net<>::Forward()
@ 0x7f205b2a4f8a caffe::Solver<>::Test()
@ 0x7f205b2a580e caffe::Solver<>::TestAll()
@ 0x7f205b2a7ddd caffe::Solver<>::Step()
@ 0x7f205b2a879e caffe::Solver<>::Solve()
@ 0x40af75 train()
@ 0x4086bc main
@ 0x7f20597a4f45 (unknown)
@ 0x408e8d (unknown)
@ (nil) (unknown)
[/snip]

**Ways to Resolve?**
Maybe somewhere I have to change a setting to accommodate the GPU architecture?  I was also assuming (probably incorrectly) that AWS DIGITS 6 already had the NVIDIA driver, cuDNN, and NVcaffe, but perhaps I have to follow some of the steps earlier in the tutorial to install them? 

Thanks for your help!

dusty-nv, 2017-12-01:
Hi alamesa, you are correct that 'invalid device function' typically means the correct GPU architecture hasn't been set somewhere (in this case probably NVcaffe).

Have you been able to run any DIGITS examples, or just seeing this problem with DetectNet?   Sorry, I haven't installed this AMI myself yet.

Also, you may eventually run into the issue that TensorRT currently needs to use models trained by NVcaffe-0.15 due to some BN-normalization parsing issue.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

alamesa, 2017-12-11:
Hi @dusty-nv, thanks for the suggestions!  DetectNet is the only example that I have tried on DIGITS so far. 

I kept running into dependency issues with the DIGITS 6 AMI, so instead I moved to the NVIDIA GPU Cloud and used an nvidia-docker container with DIGITS 6 there.  This allowed me to train the DetectNet model successfully with the dog image. 

However, once I downloaded the model and tried to run inference on the Jetson TX2, I'm running into problems very similar to issues #84 and #99.  My nvidia-docker container uses Caffe version 0.16.4, yet it seems I should be using 0.15.  Is there any way to specify the use of Caffe 0.15 on the NVIDIA GPU Cloud? 

My versions on NVIDIA GPU Cloud: 
[image]