dusty-nv/jetson-inference #170: TensorRT 3 and CuDNN 7 compatibility issues

jtrovato, 2017-12-06:
I recently upgrade to Tensor RT 3 and cudnn 7 in hopes of improving the latency of my inference. Unfortunately, I am running into the following runtime error: 

detectnet-console: cudnnEngine.cpp:639: bool nvinfer1::cudnn::Engine::deserialize(const void*, std::size_t, nvinfer1::IPluginFactory*): Assertion `size >= bsize && "Mismatch between allocated memory size and expected size of serialized engine."' failed.

I tried this with my custom net and with the included coco nets. After updating I re-made the build folder and I am still getting this error. Any help/advice is much appreciated!

dusty-nv, 2017-12-06:
Hi Joe, does the imagenet-console work or produces a similar error?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

jtrovato, 2017-12-07:
Hi, I have reverted to TensorRT 2 to get back up an running but I can test that when I revisit this. 

I also compiled this repo on an x86 computer and installed Tensor RT 2.1 alongside CUDA 8 and CuDNN 7 and I am getting the same error. To be fair, this is the first time I've been trying to run this code on an x86 architecture.

dusty-nv, 2017-12-07:
Hi, latest JetPack 3.2 Developer Preview has been released for TX2, it includes automated install of TensorRT 3.0 RC2 alongside CUDA 9.0 and cuDNN7:

https://devtalk.nvidia.com/default/topic/1027301/jetson-tx2/jetpack-3-2-mdash-l4t-r28-2-developer-preview-for-jetson-tx2/

jtrovato, 2017-12-07:
Awesome! If I don't want to reflash my kernel, can I just install the target software I need: CUDA, CuDNN, TensorRT? I know I can, but will there be any kind of compatibility issues? Do I need to delete my old versions of those packages?

psyhtest, 2017-12-08:
I'm getting a similar error both on:
- GTX 1080 (CUDA 9.0.176, cuDNN 7.0.5, TensorRT 3.0.1):
```
tensorrt-time: cudnnEngine.cpp:640: bool nvinfer1::cudnn::Engine::deserialize(const void*, std::size_t, nvinfer1::IPluginFactory*): Assertion `size >= bsize && "Mismatch between allocated memory size and expected size of serialized engine."' failed.
```
and
- TX1 with JetPack 3.1 (TensorRT 2.1.2, cuDNN 6.0.21):
```
tensorrt-time: cudnnEngine.cpp:605: bool nvinfer1::cudnn::Engine::deserialize(const void*, std::size_t, nvinfer1::IPluginFactory*): Assertion `size >= bsize && "Mismatch between allocated memory size and expected size of serialized engine."' failed.
```
(just the line numbers are different).

The test program is [`tensorrt-time`](https://github.com/dividiti/ck-tensorrt/blob/master/program/tensorrt-time/tensorrt-time.cpp) from [CK-TensorRT](https://github.com/dividiti/ck-tensorrt). I will provide full instructions how to reproduce a bit later.

By the way, `imagenet-console` fails to run too e.g. on GTX 1080:
```
$ export LD_LIBRARY_PATH=/usr/local/cuda-9.0.176/lib64/:/usr/local/TensorRT-3.0.1/lib
$ ./imagenet-console orange_1.jpg                                                                                                                                                     
imagenet-console
  args (2):  0 [./imagenet-console]  1 [orange_1.jpg]  

imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet
```

Any ideas?

psyhtest, 2017-12-11:
I've played a bit more with `imagenet-console`. After placing / linking `googlenet` files into `bin/networks:
```
$ ls -la networks/*
lrwxrwxrwx 1 anton dvdt       72 Dec 11 18:39 networks/bvlc_googlenet.caffemodel -> /home/anton/CK_TOOLS/caffemodel-bvlc-googlenet/bvlc_googlenet.caffemodel
-rw-rw-r-- 1 anton dvdt 28064352 Dec 11 18:41 networks/bvlc_googlenet.caffemodel.2.tensorcache
-rw-rw-r-- 1 anton dvdt    42924 Dec 11 12:07 networks/detectnet.prototxt
-rw-rw-r-- 1 anton dvdt    35815 Dec 11 18:41 networks/googlenet.prototxt
-rw-rw-r-- 1 anton dvdt    31675 Dec 11 12:07 networks/ilsvrc12_synset_words.txt
```
I get:
```
$ export LD_LIBRARY_PATH=/usr/local/cuda-9.0.176/lib64/:/usr/local/TensorRT-3.0.1/lib
$ ./imagenet-console granny_smith_0.jpg granny_smith.output                                                                                                                          
imagenet-console
  args (3):  0 [./imagenet-console]  1 [granny_smith_0.jpg]  2 [granny_smith.output]  

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   2

[GIE]  TensorRT version 3.0, build 3001
[GIE]  attempting to open cache file networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform does not have FP16 support.
[GIE]  loading networks/googlenet.prototxt networks/bvlc_googlenet.caffemodel
[GIE]  retrieved output tensor 'prob'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  completed building CUDA engine
[GIE]  network profiling complete, writing cache to networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  completed writing cache to networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  networks/bvlc_googlenet.caffemodel loaded
[GIE]  CUDA engine context initialized with 2 bindings
[GIE]  networks/bvlc_googlenet.caffemodel input  binding index:  0
[GIE]  networks/bvlc_googlenet.caffemodel input  dims (b=2 c=3 h=224 w=224) size=1204224
[cuda]  cudaAllocMapped 1204224 bytes, CPU 0x10210c00000 GPU 0x10210c00000
[GIE]  networks/bvlc_googlenet.caffemodel output 0 prob  binding index:  1
[GIE]  networks/bvlc_googlenet.caffemodel output 0 prob  dims (b=2 c=1000 h=1 w=1) size=8000
[cuda]  cudaAllocMapped 8000 bytes, CPU 0x10210e00000 GPU 0x10210e00000
networks/bvlc_googlenet.caffemodel initialized.
[GIE]  networks/bvlc_googlenet.caffemodel loaded
imageNet -- loaded 1000 class info entries
networks/bvlc_googlenet.caffemodel initialized.
loaded image  granny_smith_0.jpg  (400 x 400)  2560000 bytes
[cuda]  cudaAllocMapped 2560000 bytes, CPU 0x10211000000 GPU 0x10211000000
[cuda]   cudaGetLastError()
[cuda]      no kernel image is available for execution on the device (error 48) (hex 0x30)
[cuda]      /home/anton/CK_TOOLS/lib-jetson-inference-nvidia-trunk2-gcc-6.3.0-compiler.cuda-9.0.176-linux-64/src/imageNet.cu:115
[cuda]   cudaPreImageNetMean((float4*)rgba, width, height, mInputCUDA, mWidth, mHeight, make_float3(104.0069879317889f, 116.66876761696767f, 122.6789143406786f))
[cuda]      no kernel image is available for execution on the device (error 48) (hex 0x30)
[cuda]      /home/anton/CK_TOOLS/lib-jetson-inference-nvidia-trunk2-gcc-6.3.0-compiler.cuda-9.0.176-linux-64/src/imageNet.cpp:296
imageNet::Classify() -- cudaPreImageNetMean failed
imagenet-console:  failed to classify 'granny_smith_0.jpg'  (result=-1)

shutting down...
```

dusty-nv, 2017-12-11:
What GPU are you using?

jetson-inference is officially supported on Jetson TX1/TX2, but with some tweaks can be compiled for discrete GPUs.

It looks like you need to add your GPU compute_/sm_ to CMakeLists.txt:37:

https://github.com/dusty-nv/jetson-inference/blob/8ed492bfdc9e1b98f7711cd5ae224ce588635656/CMakeLists.txt#L37

psyhtest, 2017-12-11:
Thanks @dusty-nv! This worked for both on TX1 and GTX1080:
```
$ ./imagenet-console ./granny_smith_0.jpg ./granny_smith_out.jpg
...
class 0948 - 0.999714  (Granny Smith)
imagenet-console:  './granny_smith_0.jpg' -> 99.97144% class #948 (Granny Smith)
loaded image  fontmapA.png  (256 x 512)  2097152 bytes
[cuda]  cudaAllocMapped 2097152 bytes, CPU 0x10211400000 GPU 0x10211400000
[cuda]  cudaAllocMapped 8192 bytes, CPU 0x10210e02000 GPU 0x10210e02000
imagenet-console:  attempting to save output image to './granny_smith_out.jpg'
imagenet-console:  completed saving './granny_smith_out.jpg'
```
(The output file name must be given _and_ must end with `.jpg` apparently.)

While this issue is out of the way, the original problem that @jtrovato and I had still stands...

psyhtest, 2017-12-11:
My problem can be reproduced as follows:
```
$ sudo pip install ck
$ ck pull repo:ck-tensorrt --url=https://github.com/dividiti/ck-tensorrt
$ ck compile program:tensorrt-test
$ ck run program:tensorrt-test --cmd_key=default
```
- Select any model e.g. `caffemodel-bvlc-googlenet` and any image e.g. `cropped_panda.jpg`
- See the program fail:
```
executing code ...
./tmp-OllMFB.sh: line 32:  3853 Segmentation fault      ./tensorrt-test /home/anton/CK_REPOS/ctuning-datasets-min/dataset/image-jpeg-dnn-cropped-panda/cropped_panda.jpg > stdout.log 2> stderr.log
```
- Check the program's `stderr`:
$ cat `ck find program:tensorrt-time`/tmp/stderr.log
tensorrt-time: cudnnEngine.cpp:605: bool nvinfer1::cudnn::Engine::deserialize(const void*, std::size_t, nvinfer1::IPluginFactory*): Assertion `size >= bsize && "Mismatch between allocated memory size and expected size of serialized engine."' failed.
```

jtrovato, 2017-12-13:
@dusty-nv I have updated my Jetson TX2 with the new Jetpack 3.2(Tensor RT 3, CUDA 9, CuDNN 7). Examples in jetson-inference are running fine but my custom net still produces the following error:

`detectnet-console: cudnnEngine.cpp:639: bool nvinfer1::cudnn::Engine::deserialize(const void*, std::size_t, nvinfer1::IPluginFactory*): Assertion size >= bsize && "Mismatch between allocated memory size and expected size of serialized engine."' failed.`

Do I need to retrain my custom network with the newest version of CUDA and CuDNN on my host computer? Is that why the examples are working and my custom net does not?

psyhtest, 2017-12-19:
@jtrovato We found that our issues were due to serialization API changes since TensorRT 1.0 EA. Please take a took at the fix [here](https://github.com/dividiti/ck-tensorrt/pull/10/files). Hope this helps.

jtrovato, 2018-01-08:
@dusty-nv How do you train your nets on DIGITS? Do I need to update CUDA, CuDN, and NVCaffe on the Host for my custom nets to be compatible with TensorRT 3? I can successfully run the examples but my custom net fails.

dusty-nv, 2018-01-08:
Hi Joe, are you using nvcaffe-0.15 branch?

Thanks,
Dustin

jtrovato, 2018-01-09:
Yes! I am on the nvcaffe-0.15 branch

closed 2023-04-27 by dusty-nv