dusty-nv/jetson-inference #171: TensorRT 3.0 SSD ERROR

chenzhi1992, 2017-12-07:
I'm trying to build the objection detection framework SSD with TensorRT 3.0 plugin api, the DetectionOutput layer was created by the create function in the NVIDIA plugin library createSSDDetectionOutputPlugin. But it does return a error message "Plugin layer output count is not equal to caffe output count".When I remove the DetectionOutput layer in the prototxt file, it can run and I can get the output of the  mbox_loc, mbox_conf_flatten, mbox_priorbox layer.
 Can you help me ? Thank you!

dusty-nv, 2017-12-12:
Hi @chenzhi1992 , have you seen this example of using TensorRT plugin API?

https://github.com/AastaNV/Face-Recognition

twmht, 2017-12-13:
@chenzhi1992 

Where you find SSD repo (or github)  for TensorRT 3.0?

chenzhi1992, 2017-12-15:
@dusty-nv Yeah， I have seen it. In tensorrt3.0, the detectout layer of SSD has been implemented ， and when I used this api， I meet the error. Thanks!

chenzhi1992, 2017-12-15:
@twmht Please download TensorRT3.0 (https://developer.nvidia.com/nvidia-tensorrt3rc-download).And in TensorRT3.0 api, you can find some SSD plugin layer.

twmht, 2017-12-15:
@chenzhi1992 

Have you ever tried INT8 for faster-rcnn in TensorRT?

chenzhi1992, 2017-12-15:
@twmht I dont tried it， because I haven't touched that much. Maybe I'll study it and try INT8 for SSD in TensorRT later . Thanks.

twmht, 2017-12-15:
I have tried faster-rcnn before, but it did not work. The final mAP is very low.

If you have any good news, please let me know.

thanks.

liuhaichao17, 2017-12-18:
Hi,@dusty-nv,@chenzhi1992, I'm also trying to build SSD with TensorRT 3.0 plugin api, and the Normalize layer was created by the function createSSDNormalizePlugin, and I got the error "could not parse layer type IPlugin" after the funtion. 
Can you help me? Thank you!

chenzhi1992, 2017-12-18:
你看下你的网络prototxt文件，是不是layer的名字不小心写错了

发自网易邮箱大师
On 12/18/2017 15:48, liuhaichao17 wrote:

Hi,@dusty-nv,@chenzhi1992, I'm also trying to build SSD with TensorRT 3.0 plugin api, and the Normalize layer was created by the function createSSDNormalizePlugin, and I got the error "could not parse layer type IPlugin" after the funtion.
Can you help me? Thank you!

liuhaichao17, 2017-12-18:
@chenzhi1992 ,prototxt文件里layer的名字没错，请问还有其他可能的原因么？谢谢

chenzhi1992, 2017-12-18:
我遇到过几次这个错误，都是名字不小心被我写错了。会不会是其他层的名字写错了，而不是只看normalize层。

发自网易邮箱大师
在2017年12月18日 16:11，liuhaichao17 写道:

@chenzhi1992 ,prototxt文件里layer的名字没错，请问还有其他可能的原因么？谢谢

liuhaichao17, 2017-12-18:
@chenzhi1992 ，嗯，我再看看其他的层，非常感谢您的回复

liuhaichao17, 2017-12-20:
@chenzhi1992 ,你好，我Normalize层用tensorrt提供的函数初始化代码如下：
const bool ACROSS_SPATIAL = false;
const bool CHANNEL_SHARED = false;
const int NORM_VALUE_NUM = 1;
const float NORM_VALUE[NORM_VALUE_NUM] = { 20.f };
const float NORM_FP = 1.00000000e-10;
		else if (!strcmp(layerName, "NormalizeCTO"))
		{
			assert(mPluginNorm == nullptr);
			assert(nbWeights == 0 && weights == nullptr);
			Weights wg{ nvinfer1::DataType::kFLOAT, NORM_VALUE, NORM_VALUE_NUM };
			mPluginNorm = std::unique_ptr<INvPlugin, decltype(nvPluginDeleter)>
				(createSSDNormalizePlugin(&wg, ACROSS_SPATIAL, CHANNEL_SHARED, NORM_FP), nvPluginDeleter);
			return mPluginNorm.get();
		}
然后报错：sample_ssd: NvPluginSSD.cu:138: virtual void nvinfer1::plugin::Normalize::configure(const nvinfer1::Dims*, int, const nvinfer1::Dims*, int, int): Assertion `mWeights.count == C' failed.
请问是我函数初始化代码的问题么，还是因为其他的原因。另外Plugin layer output count is not equal to caffe output count的问题你解决了么，谢谢。

cpg361, 2017-12-20:
createSSDNormalizePlugin 函数的 weights参数，直接用createPlugin函数的Weight参数就行了，不用自己重新定义Weight

chenzhi1992, 2017-12-20:
createSSDNormalizePlugin 函数的 weights参数，直接用createPlugin函数的Weight参数就行了，不用自己重新定义Weight

liuhaichao17, 2017-12-20:
@chenzhi1992, @cpg361,我直接用createPlugin函数的Weight参数，运行到createSSDNormalizePlugin的时候直接段错误了，是什么问题？

shuiyuelin, 2017-12-21:
@liuhaichao17 我用的自带的createSSDNormalizePlugin 是没有问题的。。。你可以把代码和prototxt normalize那部分贴详细点。

chenzhi1992, 2017-12-21:
@liuhaichao17 我用的自带的createSSDNormalizePlugin 是没有问题的。。。你可以把代码和prototxt normalize那部分贴详细点。

daniel-pp, 2018-01-08:
Any updates on this? What is the number of output dimensions for the DetectionOutput Plugin in TensorRT? Is it (1, 1, 1, 7) or (1, 1, N, 7), and if the latter, what is N assumed to be?

wahajahmad, 2018-01-18:
@chenzhi1992 
were you able to solve this error with 'detection_out' layer of SSD ?

Error:"Plugin layer output count is not equal to caffe output count"

chenzhi1992, 2018-01-18:
I removed the 'detection_out' layer, and calculated the final output through the mbox_loc, mbox_prior and mbox_conf layer output. 
You can pay attention to my github account, I will open the code soon.
Thanks

wahajahmad, 2018-01-18:
@chenzhi1992 thanks for quick response,

I have reached the same point, about to do the same for detection layer replacement

and yes i have already check that repo, keep up with that !!

szc1523, 2018-01-27:
@chenzhi1992 @wahajahmad wa。 I've solved this problem, it's because the tensorrt ssd implementation of detection output layer has TWO outputs. Therefore you should add an output blob in the prototxt file. I've built it successfully.

szc1523, 2018-01-27:
@chenzhi1992 现在我遇到一个新问题。build完之后 deserializeCudaEngine时，priorbox layer 报错。ERROR: NvPluginSSD.cu (349) - Cuda Error in ~PriorBox: 17
terminate called after throwing an instance of 'nvinfer1::CudaError'
  what():  std::exception
Aborted (core dumped)
请问您在deserializeCudaEngine有遇到这个问题么？

chenzhi1992, 2018-01-27:
可能是因为你的维度没弄好

发自网易邮箱大师
在2018年01月27日 20:40，szc1523 写道:

@chenzhi1992 现在我遇到一个新问题。build完之后 deserializeCudaEngine时，priorbox layer 报错。ERROR: NvPluginSSD.cu (349) - Cuda Error in ~PriorBox: 17
terminate called after throwing an instance of 'nvinfer1::CudaError'
what(): std::exception
Aborted (core dumped)
请问您在deserializeCudaEngine有遇到这个问题么？

szc1523, 2018-01-27:
嗯你是用的官方的priorbox plugin，然后没有问题么？

szc1523, 2018-01-27:
你是用的官方的prior box layer。然后deserializeCudaEngine 不报错嘛? @chenzhi1992

chenzhi1992, 2018-01-27:
没有哎

发自网易邮箱大师

wahajahmad, 2018-01-28:
thanks @szc1523 
Let me try this

wahajahmad, 2018-01-31:
@szc1523 this works man !

However without the 'detection_out' layer I was getting ~106ms for inference at batch size = 1
and after adding this layer inference time drastically exceeds to ~125ms at same batch size
(at MaxN mode and Max Jetson clocks)

Are you having the same results ??

szc1523, 2018-02-26:
@wahajahmad this layer is slow. So this time consumption is expected.

hexiangquan, 2018-03-28:
@szc1523  you get right result ?

szc1523, 2018-03-29:
@hexiangquan yes.

hexiangquan, 2018-04-03:
tx2 i just get  43fps

发自我的 iPhone

wahajahmad, 2018-05-17:
@hexiangquan woah 43 Fps ? what was the netowrk you used ?

aditbhrgv, 2018-06-18:
Does anyone know how to solve the error  "could not parse layer type IPlugin" while running ./giexec engine in TensorRT? 
THanks

jealous1989, 2018-07-11:
@szc1523 @wahajahmad @chenzhi1992  fix detectionout layers' bug to add two output? did you find other error? when set tensorrt3's markOutput layer to "detectionout", it shows  error "NvPluginSSD.cu:587: virtual void nvinfer1::plugin::DetectionOutput::configure(const nvinfer1::Dims*, int, const nvinfer1::Dims*, int, int): Assertion `numPriors*numLocClasses*4 == inputDims[0].d[0]' failed" .  
and two output ,which one that I can get the detection result?

z13974509906, 2018-07-13:
@jealous1989 have u solve the problem? I met the same error, which model are u using? I met it with the mobile-ssd

jealous1989, 2018-07-16:
@z13974509906 me two, I used mobile_v2 as base_net.
 Now I add two output to detectionout layer in prototxt. And also get the two outputs in my code.  

If successed, I will reply u later.

JingliangGao, 2018-07-17:
When I plan to accelerate the Mobilenet-SSD network , here comes a trouble .

In Flatten -- inputs[0].d[0] -> 19 -- inputs[0].d[1] -> 19 -- inputs[0].d[2] -> 12
In Flatten -- inputs[0].d[0] -> 19 -- inputs[0].d[1] -> 19 -- inputs[0].d[2] -> 63
In Flatten -- inputs[0].d[0] -> 10 -- inputs[0].d[1] -> 10 -- inputs[0].d[2] -> 24
In Flatten -- inputs[0].d[0] -> 10 -- inputs[0].d[1] -> 10 -- inputs[0].d[2] -> 126
In Flatten -- inputs[0].d[0] -> 5 -- inputs[0].d[1] -> 5 -- inputs[0].d[2] -> 24
In Flatten -- inputs[0].d[0] -> 5 -- inputs[0].d[1] -> 5 -- inputs[0].d[2] -> 126
In Flatten -- inputs[0].d[0] -> 3 -- inputs[0].d[1] -> 3 -- inputs[0].d[2] -> 24
In Flatten -- inputs[0].d[0] -> 3 -- inputs[0].d[1] -> 3 -- inputs[0].d[2] -> 126
In Flatten -- inputs[0].d[0] -> 2 -- inputs[0].d[1] -> 2 -- inputs[0].d[2] -> 24
In Flatten -- inputs[0].d[0] -> 2 -- inputs[0].d[1] -> 2 -- inputs[0].d[2] -> 126
In Flatten -- inputs[0].d[0] -> 1 -- inputs[0].d[1] -> 1 -- inputs[0].d[2] -> 24
In Flatten -- inputs[0].d[0] -> 1 -- inputs[0].d[1] -> 1 -- inputs[0].d[2] -> 126
In Reshape -- inputs[0].d[0] -> 40257 -- inputs[0].d[1] -> 1 -- inputs[0].d[2] -> 1 -- OutC -> 21
In SoftMax -- inputs[0].d[0] -> 1917 -- inputs[0].d[1] -> 21 -- inputs[0].d[2] -> 1
In Flatten -- inputs[0].d[0] -> 1917 -- inputs[0].d[1] -> 21 -- inputs[0].d[2] -> 1
inference: NvPluginSSD.cu:431: virtual void nvinfer1::plugin::PriorBox::configure(const nvinfer1::Dims*, int, const nvinfer1::Dims*, int, int): Assertion `H == inputDims[0].d[1]' failed.

Who could tell me how to solve this problem or give a little advice? Thanks~

myih, 2018-08-11:
@jealous1989 @z13974509906 I have the same error 
` "NvPluginSSD.cu:587: virtual void nvinfer1::plugin::DetectionOutput::configure(const nvinfer1::Dims*, int, const nvinfer1::Dims*, int, int): Assertion 'numPriors*numLocClasses*4 == inputDims[param.inputOrder[0]].d[0]' failed."`
Any update? Thanks

myih, 2018-08-13:
@jealous1989 @z13974509906 
I got the same error but turns out it was caused by wrong name of plugin layer. 
I did found something related, if you are using TensorRT4's createSSDDetectionOutputPlugin, you have to specify inputOrder {0,1,2} to fix this error, which is not needed in TensorRT3. 

But after fixing all the errors, my detection result is incorrect. 
If anyone manage to run mobilenet-ssd with TensorRT, it would be appreciated if you can provide some direction.

niuhope, 2018-08-22:
@myih @dusty-nv 
I'm using TensorRT4's DetectionOutputPlugin, and have the same problem. I can only get one detection result and the detected object of multiple runs is different with a small nmsThreshold(my caffemodel's nmsThreshold is 0.3). When I set a lagre nmsThreshold(like 0.9) it seens to be correct(at least it output multiple correct results), but I don't understand why.

myih, 2018-08-22:
@niuhope Are you using a TX2? Your problem sounds similar to mine on the TX2.
I got correct result on 1080ti machine, but when ported to TX2 the detection rate is much lower with the same threshold.

niuhope, 2018-08-22:
@myih Yes, I'm using a TX2, but I got wrong result on 1080 machine too. I modify prototxt as follows: 1. change flatten layer to reshape layer; 2. use permute layer to transpose conf for using softmax layer with axis = 1. I can get correct result on caffe with modified prototxt.

niuhope, 2018-08-31:
@myih I implement a detect out layer by myself and get correct result

ujsyehao, 2018-11-28:
@niuhope Hi, Can you show your detection_output layer code?

ujsyehao, 2018-11-28:
@chenzhi1992 Hi, Now tensorRT3.0 has no ssd plugin layer code. Can you send me it?

closed 2023-04-27 by dusty-nv