dusty-nv/jetson-inference #172: TensorRT builder->setMaxBatchSize(maxBatchSize); question

rockkingjy, 2017-12-09:
I've seen in the TensorRT developer guide document that there is a:

builder->setMaxBatchSize(maxBatchSize);

with explanation:
‣ maxBatchSize is the size for which the engine will be tuned. At execution time, smaller batches may be used, but not larger.

But I am not quite clear of this parameter, could anyone help me to clarify this parameter?

What does BatchSize here means? Is it the same meaning of batch size in deep neural work training? Or other meanings?

rockkingjy, 2017-12-09:
It's frequently used here, for example in imagines.h:
```
static imageNet* Create( NetworkType networkType=GOOGLENET, uint32_t maxBatchSize=2 );
```
with explanation:
```
 @param maxBatchSize The maximum batch size that the network will support and be optimized for.
```
But I'm not quite clear of this, could anyone help me?

dusty-nv, 2017-12-12:
Hi @rockkingjy , yes it is the same meaning as batch size in neural network training.  The batch size indicates how many images are processed at once.  It can be up to 128 with GoogleNet or AlexNet, for example.

Depending on network, TensorRT can accept a variable batch size at runtime, so it can process different batch sizes.  However to do this it needs to know the maximum batch size you wish to support, so it can allocate the resources required in advance.

closed 2017-12-12 by dusty-nv

rockkingjy, 2017-12-14:
Hi @dusty-nv , thanks for your answer. I'm trying to do is to classify more than one image at the same time to save running time. I do this by change the code in imageNet.cpp:
```
	if( CUDA_FAILED(cudaPreImageNetMean((float4*)rgba, width, height, mInputCUDA, mWidth, mHeight,make_float3(104.0069879317889f, 116.66876761696767f, 122.6789143406786f))) )
	{
		printf("imageNet::Classify() -- cudaPreImageNetMean failed\n");
		return -1;
	}

	if( CUDA_FAILED(cudaPreImageNetMean((float4*)rgba1, width1, height1, mInputCUDA+602112, mWidth, mHeight,make_float3(104.0069879317889f, 116.66876761696767f, 122.6789143406786f))) )
	{
		printf("imageNet::Classify() -- cudaPreImageNetMean failed\n");
		return -1;
	}

	// process with GIE
	void* inferenceBuffers[] = { mInputCUDA, mOutputs[0].CUDA };
	mContext->execute(MAXBATCHSIZE, inferenceBuffers);
```
Here, in:
```
if( CUDA_FAILED(cudaPreImageNetMean((float4*)rgba1, width1, height1, mInputCUDA+602112, 
```
602112=224*224*3*4 is the memory for one input image, I put the second image after the first one. And here:
```
mContext->execute(MAXBATCHSIZE, inferenceBuffers);
```
I've predefined:
```
#define MAXBATCHSIZE 2
```
in imageNet.h. And I've changed all the other code files related to maxBatchSize to this MAXBATCHSIZE.

But unfortunately, it seems not returning the result for the second input image, just working for the first image.