dusty-nv/jetson-inference #175: OpenGL segfault in jetson-inference examples

ZeroCool11000, 2017-12-19:
I am always getting segfaults if i want to use the camera not console from those examples , so if try: 

nvidia@tegra-ubuntu:~/Downloads/jetson-inference/build/aarch64/bin$ ./detectnet-camera
detectnet-camera
  args (1):  0 [./detectnet-camera]

(gst-plugin-scanner:6086): GStreamer-WARNING **: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstclutter-3.0.so': /usr/lib/aarch64-linux-gnu/libgbm.so.1: undefined symbol: drmGetDevice2
[gstreamer] initialized gstreamer, version 1.8.3.0
[gstreamer] gstreamer decoder pipeline string:
nvcamerasrc fpsRange="30.0 30.0" ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12 ! nvvidconv flip-method=0 ! video/x-raw ! appsink name=mysink

detectnet-camera:  successfully initialized video device
    width:  1280
   height:  720
    depth:  12 (bpp)

detectNet -- loading detection network model from:
          -- prototxt    networks/ped-100/deploy.prototxt
          -- model       networks/ped-100/snapshot_iter_70800.caffemodel
          -- input_blob  'data'
          -- output_cvg  'coverage'
          -- output_bbox 'bboxes'
          -- mean_pixel  0.000000
          -- threshold   0.500000
          -- batch_size  2

[GIE]  TensorRT version 2.1, build 2102
[GIE]  attempting to open cache file networks/ped-100/snapshot_iter_70800.caffemodel.2.tensorcache
[GIE]  loading network profile from cache... networks/ped-100/snapshot_iter_70800.caffemodel.2.tensorcache
[GIE]  platform has FP16 support.
[GIE]  networks/ped-100/snapshot_iter_70800.caffemodel loaded
[GIE]  CUDA engine context initialized with 3 bindings
[GIE]  networks/ped-100/snapshot_iter_70800.caffemodel input  binding index:  0
[GIE]  networks/ped-100/snapshot_iter_70800.caffemodel input  dims (b=2 c=3 h=512 w=1024) size=12582912
[cuda]  cudaAllocMapped 12582912 bytes, CPU 0x102a00000 GPU 0x102a00000
[GIE]  networks/ped-100/snapshot_iter_70800.caffemodel output 0 coverage  binding index:  1
[GIE]  networks/ped-100/snapshot_iter_70800.caffemodel output 0 coverage  dims (b=2 c=1 h=32 w=64) size=16384
[cuda]  cudaAllocMapped 16384 bytes, CPU 0x103600000 GPU 0x103600000
[GIE]  networks/ped-100/snapshot_iter_70800.caffemodel output 1 bboxes  binding index:  2
[GIE]  networks/ped-100/snapshot_iter_70800.caffemodel output 1 bboxes  dims (b=2 c=4 h=32 w=64) size=65536
[cuda]  cudaAllocMapped 65536 bytes, CPU 0x103800000 GPU 0x103800000
networks/ped-100/snapshot_iter_70800.caffemodel initialized.
[cuda]  cudaAllocMapped 16 bytes, CPU 0x103a00000 GPU 0x103a00000
maximum bounding boxes:  8192
[cuda]  cudaAllocMapped 131072 bytes, CPU 0x103c00000 GPU 0x103c00000
[cuda]  cudaAllocMapped 32768 bytes, CPU 0x103810000 GPU 0x103810000
default X screen 0:   1920 x 1080
[OpenGL]  glDisplay display window initialized
[OpenGL]   creating 1280x720 texture
X Error of failed request:  BadRequest (invalid request code or no such operation)
  Major opcode of failed request:  146 (GLX)
  Minor opcode of failed request:  187 ()
  Serial number of failed request:  102
  Current serial number in output stream:  102

I guess this is perhabs because of opengl, i was also able to find the problem more precisely in function bool glTexture::init( uint32_t width, uint32_t height, uint32_t format, void* data ).  Then the segfault was dropped during this command GL(glGenBuffers(1, &dma)); 

Thx for help.

dusty-nv, 2017-12-19:
Are you running this on Jetson TX1 or TX2?

Do you have an HDMI monitor attached?  If so, what is the display resolution?  Thanks.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

ZeroCool11000, 2017-12-19:
Ah sorry i just forgot to say i am using TX2 and remote console atm, the client uses mobaxterm, i was also able to run a simple opengl program on it which is called glxgears, which was no problem.

Thx.

dusty-nv, 2017-12-19:
I see, yes unfortunately GL tunneling will not work for this texturing and DMA-intensive application with the camera video.    You can still run detectnet-camera in console mode though by setting the glDisplay pointer to NULL (instead of creating the window) here:

https://github.com/dusty-nv/jetson-inference/blob/8ed492bfdc9e1b98f7711cd5ae224ce588635656/detectnet-camera/detectnet-camera.cpp#L134

ZeroCool11000, 2017-12-19:
Thank you very much, so i have to save a video anywhere or i have to use a monitor. 

But best regards and thank you.

closed 2023-04-27 by dusty-nv