dusty-nv/jetson-inference #187: Why is cuda used to convert to rgba and not nvvidconf?

plieningerweb, 2018-01-18:
The detectnet example uses https://github.com/dusty-nv/jetson-inference/blob/master/util/camera/gstCamera.cpp  and converts rgb or nv12 to rgba in cuda.  But why not directly output rgba from the gstreamer pipeline? (..., format=Rgba) It seems I am missing why you chose this way...

Performance reasons? Quality? Or why?

dusty-nv, 2018-01-30:
Hi plieningerweb, getting the data out of gstreamer involves using `appsink` element which has an extra memory copy.  Hence it was done to save bandwidth.   However it would interesting to try it the way you highlighted.

imrisaac, 2020-01-03:
I have been fighting to do a very similar thing. Leveraging the ISP for the the NV12->RGBA conversion. 
using this pipeline 

    ss << "nvarguscamerasrc "
              "sensor-id=" << mSensorCSI << " "
              "maxperf=true ! "
            "video/x-raw(memory:NVMM), "
              "width=(int)" << mWidth << ", "
              "height=(int)" << mHeight << ", "
              "framerate=60/1, "
              "format=(string)NV12 ! "
            "nvvidconv "
              "flip-method=" << flipMethod << " ! "
            "video/x-raw, "
              "format=(string)RGBA ! ";

But I can not seem to correctly access the RGBA image. I always end up with a very distorted image from what looks like an incorrect colorspace conversion.

closed 2023-04-27 by dusty-nv