dusty-nv/jetson-inference #190: TX2/JetPack 3.2 - CMakePreBuild.sh - broken 'glew-utils' package is breaking the cmake build

jeremyolsen, 2018-01-29:
I'm taking the Udacity Robotics ND and this is the sample project for one of the labs.  I'm using the TX2 with JetPack 3.2.  The following line in the CMakePreBuild.sh is breaking b/c the 'glew-utils' package is not being found:
```
sudo apt-get install -y libqt4-dev qt4-dev-tools libglew-dev glew-utils libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev libglib2.0-dev
```

At the end of the build, it results in this error:
```
CMake Error at /usr/share/cmake-3.5/Modules/FindQt4.cmake:1326 (message):
  Found unsuitable Qt version "" from NOTFOUND, this code requires Qt 4.x
-- Configuring incomplete, errors occurred!
```

When I try to install `glew-utils` separately, I get the following error:
```
Package glew-utils is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'glew-utils' has no installation candidate
```

Even if I install the rest of the QT packages manually ignoring the 'glew-utils', there are still errors with the build.  Do I need to go to JP3.1 to fix this issue?  That's a considerable downgrade of both the host and TX2 to have to accomplish this.

dusty-nv, 2018-01-29:
Hi Jeremy, these errors seem strange – perhaps is an environment issue, or did JetPack not complete the post-install steps successfully?   You are running this from the Jetson, correct?  Are you able to run CUDA samples or “nvcc --version”?

If you re-run the cmake ../ step after installing qt4-packages, does CMake pick up Qt and stop throwing the Qt NOTFOUND error?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

jeremyolsen, 2018-01-30:
Hi @dusty-nv.  I've flashed my TX2 4 separate times now thinking it was an environment issue.  The host is a freshly installed 16.04LTS.  I'm installing JP3.2 'as-is' as a 'Full' install and every package in the list has an 'install' selected.  And I'm verifying that at the end of the flashing, the x-term window is finished and closes before clicking 'Finish".  Everything looks happy as far as I can tell.

Back on the TX2, I just tried running 'nvcc --version' from a new terminal, assuming it's already in the system path and getting 'command not found'.  I'm assuming that this means CUDA didn't install correctly.  I'm also not finding any of the samples on the TX2 that are listed at the end of the installation guide: http://docs.nvidia.com/jetpack-l4t/index.html#developertools/mobile/jetpack/l4t/3.2/jetpack_l4t_install.htm  

So what am I doing wrong here?

dusty-nv, 2018-01-30:
Hmm if it’s not finding nvcc, that does indeed likely mean that CUDA toolkit wasn’t installed properly, and most likely pointing to an issue during the JetPack stage.  During flashing with JetPack, you need the Jetson connected to host PC via both micro USB cable and Ethernet.   The Ethernet is so that JetPack can install the packages like CUDA, cuDNN, and others over the network via SSH.  The Ethernet connection can be via a router or a direct connection to a spare NIC port on your host PC.

jeremyolsen, 2018-01-30:
@dusty-nv - as in physically connect CAT5's and not using WIFI.  Makes sense...I kind of assumed it as using the latter.  Sounds like I've been pointed in the right direction.  Thanks for looking at this!

dusty-nv, 2018-01-30:
Aha, if you were using WiFi, users have reported intermittent problems using JetPack over WiFi (not to mention the added complexity of making sure your Jetson is connected to WiFi network before JetPack begins that stage).  The officially supported way is done by connecting the Jetson and PC with a physical CAT5 (either directly or via router/switch).

jeremyolsen, 2018-01-30:
@dusty-nv - My problems have been solved.  Thanks!

I'm now finding all of the niche posts about using wifi for flashing/updating.  My question is if there a manual way to run those post-install scripts?  I mean, from my observations from the console log, that this is performing a convenience step that is downloading source/compiling and installing packages located on NVIDIA systems after the OS boots up.  Looks like a lot to manage, but  if I already had the code/packages on a drive and flashing in the field is necessary, then this is not a bad option to have 'just in case'.  

Is there a script I could look at and maybe modify to suit my purposes?

closed 2018-01-30 by jeremyolsen

VirajDeshwal, 2019-04-17:
I had the same problem and it was occurring because i deleted the jetson-interface directory and I was trying to flash jetson again and again with  old-download and installed libraries on jetpack. Due to this i was having the issue of qt4 and cmake. I uninstall the old downloaded libraries in jetpack and downloaded a fresh copy and flashed the jetson again. It fixed my problem.