dusty-nv/jetson-inference #192: how can i handle a network needs different image size

qingzew, 2018-02-03:
I have seen the tensorrt sample about faster-rcnn, it has a fixed image size, so when I want to use different image, I have two questions:

1.tensorrt will do the optimizations according to network's input layer, or the input shape:
```
layer {
  name: "data"
  type: "Input"
  top: "data"
  input_param { shape: { dim: 64 dim: 1 dim: 28 dim: 28 } }
}
```

2.when doing inference,  can  I `CudaMalloc` a memory which has a different size with input layer's shape param

is there a demo about variable size of input image size?

dusty-nv, 2018-02-10:
Hi qingzew, the `imageNet`, `detectNet`, and `segNet` classes automatically read the input size dimensions from TensorRT from the input layer, and perform re-scaling on the raw image/video that you feed to the API, so you shouldn't have to worry about this too much.

qingzew, 2018-02-23:
Hi @dusty-nv do you mean 'cudaPreImageNet()', but sometimes a net have a different size input to get a better precision, when I resize it to the same size,  I will have bad result, so any good solutions?

romzie, 2018-06-28:
@qingzew have you found a solution or a snippet to help?

qingzew, 2018-06-29:
no, maybe a better idea is to set a max size

romzie, 2018-06-29:
Hi qingzew, 
If you set a max size (which I did), do you know how to retreive the result knowing its dimensions?
I have convolutions so output size is smaller with a smaller input and I have trouble identifying how the result is stored in the output buffer.

qingzew, 2018-06-29:
Hi @romzie 
I don't what you are doing, detection or classification?  In fact, convolution and pooling layer reduce the dimension of the input image, but you can compute the scale of that, for example, a pooling layer usually has a scale 1/2, if you has point (x, y), you can get (x / scale, y / scale) in origin input

romzie, 2018-06-29:
Both actually, using convolutions and pooling, so the output size is not an issue, I know it.
The issue is retrieving the result from TensorRT. I dont know how the output buffer is filled with the inference results. What I mean is that my output buffer has a maximum size so I need to truncate it to get results.

qingzew, 2018-06-30:
my idea is that when  i have max size input, i can place the input image to the center of the input blob, and padding 0, so i know the padding width and height, i can retrieve the result  now

mprat, 2018-10-12:
@dusty-nv You mentioned that the `detectNet` class automatically resizes the input images, but I can't seem to find where that code appears. Is it baked into the `cudaPreImageNetMean`?

drakorg, 2020-07-17:
> Hi qingzew, the `imageNet`, `detectNet`, and `segNet` classes automatically read the input size dimensions from TensorRT from the input layer, and perform re-scaling on the raw image/video that you feed to the API, so you shouldn't have to worry about this too much. For example, you can push an HD image into the API and it will automatically downscale it to match the network input size.

Hi, sorry to bring this back up after so much time, but I think your explanation begs the question about aspect ratios. I'd assume that the way these classes fit a bigger image into the required network size would be by resizing without keeping aspect ratio (otherwise it would have to crop some part and the decision of what part to crop would have to be left to the class, risking making the wrong call).

If that's the case (that the images are resized by stretching) would that deformation make it more difficult for the network to make a detection? That would depend on how the network has been trained I'd guess. If they were trained on 16:9 images stretched to 1:1 aspect ratios (like 300x300) is one thing, different from a 4:3 image stretched to 1:1, different from a training set of 1:1 images (probably cropped) that would fit perfectly the input aspect ratio.

Is there any information on how the training on those networks have been made? Or at least what would be the ideal shape of the image that should be passed to each of the different networks?

Thank you.
Best regards.
Eduardo

closed 2023-04-27 by dusty-nv