dusty-nv/jetson-inference #193: about failed to load synset class descriptions

liujingxuan, 2018-02-06:
I compiled the project, the error as follow:

nvidia@tegra-ubuntu:~/jetson-inference/build/aarch64/bin$ ./imagenet-console granny_smith_1.jpg output_1.jpg
imagenet-console
  args (3):  0 [./imagenet-console]  1 [granny_smith_1.jpg]  2 [output_1.jpg]  

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   2

[GIE]  TensorRT version 2.1, build 2102
[GIE]  attempting to open cache file networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading networks/googlenet.prototxt networks/bvlc_googlenet.caffemodel
[GIE]  retrieved output tensor 'prob'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  completed building CUDA engine
[GIE]  network profiling complete, writing cache to networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  completed writing cache to networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  networks/bvlc_googlenet.caffemodel loaded
[GIE]  CUDA engine context initialized with 2 bindings
[GIE]  networks/bvlc_googlenet.caffemodel input  binding index:  0
[GIE]  networks/bvlc_googlenet.caffemodel input  dims (b=2 c=3 h=224 w=224) size=1204224
[cuda]  cudaAllocMapped 1204224 bytes, CPU 0x102a00000 GPU 0x102a00000
[GIE]  networks/bvlc_googlenet.caffemodel output 0 prob  binding index:  1
[GIE]  networks/bvlc_googlenet.caffemodel output 0 prob  dims (b=2 c=20 h=1 w=1) size=160
[cuda]  cudaAllocMapped 160 bytes, CPU 0x102c00000 GPU 0x102c00000
networks/bvlc_googlenet.caffemodel initialized.
[GIE]  networks/bvlc_googlenet.caffemodel loaded
imageNet -- loaded 1000 class info entries
imageNet -- failed to load synset class descriptions  (1000 / 1000 of 20)
imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet
nvidia@tegra-ubuntu:~/jetson-inference/build/aarch64/bin$ ./imagenet-camera googlenet
imagenet-camera
  args (2):  0 [./imagenet-camera]  1 [googlenet]  

[gstreamer] initialized gstreamer, version 1.8.3.0
[gstreamer] gstreamer decoder pipeline string:
nvcamerasrc fpsRange="30.0 30.0" ! video/x-raw(memory:NVMM), width=(int)1280, height=(int)720, format=(string)NV12 ! nvvidconv flip-method=0 ! video/x-raw ! appsink name=mysink

imagenet-camera:  successfully initialized video device
    width:  1280
   height:  720
    depth:  12 (bpp)

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   2

[GIE]  TensorRT version 2.1, build 2102
[GIE]  attempting to open cache file networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading networks/googlenet.prototxt networks/bvlc_googlenet.caffemodel
[GIE]  retrieved output tensor 'prob'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  completed building CUDA engine
[GIE]  network profiling complete, writing cache to networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  completed writing cache to networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  networks/bvlc_googlenet.caffemodel loaded
[GIE]  CUDA engine context initialized with 2 bindings
[GIE]  networks/bvlc_googlenet.caffemodel input  binding index:  0
[GIE]  networks/bvlc_googlenet.caffemodel input  dims (b=2 c=3 h=224 w=224) size=1204224
[cuda]  cudaAllocMapped 1204224 bytes, CPU 0x102a00000 GPU 0x102a00000
[GIE]  networks/bvlc_googlenet.caffemodel output 0 prob  binding index:  1
[GIE]  networks/bvlc_googlenet.caffemodel output 0 prob  dims (b=2 c=20 h=1 w=1) size=160
[cuda]  cudaAllocMapped 160 bytes, CPU 0x102c00000 GPU 0x102c00000
networks/bvlc_googlenet.caffemodel initialized.
[GIE]  networks/bvlc_googlenet.caffemodel loaded
imageNet -- loaded 1000 class info entries
imageNet -- failed to load synset class descriptions  (1000 / 1000 of 20)
imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet

dusty-nv, 2018-02-10:
Hi liujingxuan, can you inspect the text file in your local clone of the repo, at `jetson-inference/data/networks/ilsvrc12_synset_words.txt` and see if it is corrupt?

liujingxuan, 2018-02-10:
Hi dusty-nv, thanks for your reply. I check the jetson-inference/data/networks/ilsvrc12_synset_words.txt and    redownload the file,but no mistake  was found.
Then, classify an example image with the imagenet-console program:
$ ./imagenet-console orange_0.jpg output_0.jpg
But the result is the same as above.
I download many files form https://drive.google.com/drive/folders/0BwYxpotGWRNOV2d1NXNJOHA0dzQ

because I can not connect  to https://nvidia.box.com/
But all the files I've downloaded are damaged.

So ,I download bvlc_googlenet.caffemodel form http://dl.caffe.berkeleyvision.org/bvlc_googlenet.caffemodel
googlenet.prototxt from /usr/src/tensorrt/data/googlenet/googlenet.prototxt
I have read the source code of the jetson-inference , but I can not solve the error. 
Thanks for your help.

closed 2023-04-27 by dusty-nv