dusty-nv/jetson-inference #195: Overlapping Bounding Boxes Are Combined - Can This be Disabled?

ryanknopf, 2018-02-09:
Hello All,

I've trained a DetectNet Model with "Subtract Mean" = "None" on DIGITS 6.0 with Nvidia Caffe 0.15.14.  So far, following the directions to deploy this to Jetson, everything has been going as expected (well other than it taking 8-9 minutes to build the CUDA engine the first time).  However, I'm seeing very different results from detect-console than the DIGITS inference server on the same test image.  

I've noticed many other similar issues that trace back to Mean Image vs Pixel subtraction or double subtraction.  This might still be an issue, but the behavior I'm seeing is different.  In our case, it looks like all of the bounding boxes that would have overlapped with each other are being combined into one box (whereas they're not on DIGITS).  I have to assume this is related to some difference between the python clustering layer deleted from the end of deploy.prototext and the equivalent function implemented in detectNet.cpp.  Is there a clean way to modify detectNet.cpp to get identical results to the DIGITS inference tool?

Thanks for your help!
- R

ryanknopf, 2018-02-09:
So, I'm one step closer to noodling this out.  Keeping with the theory above, I've slightly modified the rectOverlap function in detectNet.cpp to make one additional check.  If the rectangle coordinates do overlap, it only returns true (in which case mergeRect will combine them) IF the area of overlap is at least 25% the area of either overlapping rectangle.  This gets me a result which is very similar to the DIGITS inference visualization, but still not quite the same.  I'm not sure how nvidia caffe normally does this clustering step in python, so it could be some difference there, or perhaps now I'm having the mean pixel subtraction issue.

From modified detectNet.cpp (for those who are looking to do the same):

inline static bool rectOverlap(const float6& r1, const float6& r2)
{
    // The rectangles do not overlap at all
    if ( r2.x > r1.z || 
         r2.z < r1.x ||
         r2.y > r1.w ||
         r2.w < r1.y ) return false;

    else 
    {
        float overlap_x = std::min(r2.z,r1.z) - std::max(r2.x,r1.x);
        float overlap_y = std::min(r2.w,r1.w) - std::max(r2.y,r1.y);
        float area_overlap = overlap_x * overlap_y;
        
        float area_r1 = (r1.z-r1.x) * (r1.w-r1.y);
        float area_r2 = (r2.z-r2.x) * (r2.w-r2.y);
        
        // The rectangles overlap by less than 75% of either's area
        if ((area_overlap < 0.25*area_r1) && 
            (area_overlap < 0.25*area_r2)) return false;
        
        // The rectangels overlap
        return true;
    }
}

dusty-nv, 2018-02-10:
Hi Ryan, you have the correct idea by adding a parameter that rejects the clustering if the overlap is less than a certain percentage of the area (in your case, 25%).   If you wanted high-quality clustering, I would look to Mean Shift Clustering algorithm, it will run slower than your approach but is very good at clustering to the center-of-mass of object while realizing that two nearby objects shouldn't be clustered.

ryanknopf, 2018-02-10:
Hi Dusty,  thanks for the pointer in the right direction!  A couple things to add before closing for any other beginners that stumble across this issue:

1) In Nvidia Caffe's clustering implementation (what DIGITS will ultimately use) it looks like the OpenCV implementation of groupRectangles() is used (as seen here in nvcaffe 0.16: [clustering.py L189](https://github.com/NVIDIA/caffe/blob/f58deb312625ac6943ec65cc279a3278ca4d17cb/python/caffe/layers/detectnet/clustering.py#L189)).
2) So, if you wanted results as similar to what your see in the DIGITS inference visualization as possible, you'd want to either reproduce groupRectangles()'s algorithmic implementation (more info here, groupRectangles wraps the more generic OpenCV clustring tools partition(): [OpenCV Docs groupRectangles](https://docs.opencv.org/3.0-beta/modules/objdetect/doc/cascade_classification.html?highlight=grouprectangles#cv2.groupRectangles)).  To get the same results you'll need to use the arguments form the python clustering layer you deleted from your deploy.prototext earlier (this is described in more detail in the comments here: [clustering.py L74](https://github.com/NVIDIA/caffe/blob/f58deb312625ac6943ec65cc279a3278ca4d17cb/python/caffe/layers/detectnet/clustering.py#L74)
3) As I understand it, one of the major disadvantages this has is it's inability to "weight" outcomes based on other knowledge about each rectangle.  For example, one of the many unclustered bounding box proposals could have a very sloppy fit around the object being detected and consequentially have a low confidence score relative to others.  Since groupRectangles()'s methodology is based purely on to what extent the the boundaries of each proposed rectangle overlap with each other (as discussed in a similar context here: [OpenCV HOG Detector Clustering](http://mccormickml.com/2013/11/07/opencv-hog-detector-result-clustering/)) that result will be given an equal weighting to one with nearly perfect confidence when clustering.  Whether or not that method is "close enough" for your purposes depends entirely on the nature of your dataset and what you're trying to do with it (as always).

closed 2018-02-15 by ryanknopf