dusty-nv/jetson-inference #200: ERROR: Did not find two valid categories

DaveBGld, 2018-02-21:
I Installed  the Digits docker image, and ran like this:

`sudo nvidia-docker run -v /data:/data/ -p 5000:5000 nvidia/digits:latest
`

Downloaded all the images, and while following all instructions in the tutorial for  creating the 12_classes dataset, I get the error in the subject.

Also, the Digits console printed the following errors:

`2018-02-21 01:15:28 [20180221-011526-b7c6] [WARNING] Parse Folder (train/val) unrecognized output: libdc1394 error: Failed to initialize libdc1394
2018-02-21 01:15:29 [20180221-011526-b7c6] [WARNING] Parse Folder (train/val): Not enough images for this category
`

Thanks in advance for your guidance and help, I am towards the end of my week two for your two days to a demo tutorial with TX2 .

Dave

AboutThatCMD, 2018-03-05:
Having a similar issue, have you made any progress past this?

DaveBGld, 2018-03-05:
I spent a lot of time on this issue and got frustrated. It appears to be that it may be either the result of wrong mappings between they host OS and the docker directory system or something related to permissions to the docker actual user that runs under Ubuntu.

In the end I just rebuilt digits locally on my machine by compiling everything from scratch and that is working

AboutThatCMD, 2018-03-05:
Interesting, I spent hours trying to work past this yesterday. By rebuilding DIGITS locally were you still able to access your Cloud GPU, or how did you work around that?

DaveBGld, 2018-03-05:
Using my local gtx 1080

dusty-nv, 2018-03-05:
Hi guys, are you able to get a command prompt from the docker instance to see if you can confirm the contents of your mounted directory?

You may want to post this to the DIGITS and/or nvidia-docker GitHubs as they have more experience than I configuring AWS/Docker.  Good luck!

Dusty

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

AboutThatCMD, 2018-03-06:
Hi Dusty,

I have posted in the DIGITS Docker support area as well. No one has replied. In my post you can see the directory contents for the folder 12_classes that we create after downloading the larger dataset. I am at a loss for what to do next. I am going to try running the DIGITS AMI instead of using docker. https://github.com/dusty-nv/jetson-inference/issues/202

dusty-nv, 2018-04-25:
Hi @AboutThatCMD , if you are still having problems, you may wish to try [NVIDIA GPU Cloud (NGC)](https://www.nvidia.com/en-us/gpu-cloud/) which automates the setup.

closed 2018-04-25 by dusty-nv

AboutThatCMD, 2018-04-25:
@dusty-nv I did use Nvidia GPU Cloud and was able to resolve this. This issue came down to what was addresses in #202 and understanding how docker allocates memory for volumes. Thank you for your help.

hbe72, 2018-11-05:
See my comment in: https://github.com/dusty-nv/jetson-inference/issues/202#issuecomment-435798621

The symbolic links are not the same in host and container side. Therefore the imagenet-subset.sh command has to be given in the container. If that command is given int the host side, the symbolic links in the container are dangling and you get the problem described above.

adfayed, 2019-06-29:
> See my comment in: [#202 (comment)](https://github.com/dusty-nv/jetson-inference/issues/202#issuecomment-435798621)
> 
> The symbolic links are not the same in host and container side. Therefore the imagenet-subset.sh command has to be given in the container. If that command is given int the host side, the symbolic links in the container are dangling and you get the problem described above.

@hbe72 How do you provide the symbolic links in the container? Could you please help guiding us with that?