dusty-nv/jetson-inference #202: Cannot Train Data in DIGITS Docker Container

AboutThatCMD, 2018-03-04:
Hi Everyone,

I am new to this and following the, 'Two Days to a Demo' path. I am on the following step:
https://github.com/dusty-nv/jetson-inference#importing-classification-dataset-into-digits

So far I have successfully created the AWS P3 instance and have downloaded the data set. I ahve also created the sub folder mentioned that is title 12_classes. I cannot get DIGITS to train this. It tells me that it did not find two valid categories. I don't necessarily believe this error due to it being a docket container. I am concerned that how I am launching the docker container is causing my issue.

Docker Command:
docker run --shm-size=1g --ulimit memlock=-1 --runtime=nvidia --name digits -d -it -p 5000:5000 -v /home/ubuntu/efs/Data/12_classes:/efs/Data/12_classes nvidia/digits

There is another example docker command I could run, but when I try alternatives I cannot access the public IP for DIGITS running on the AWS P3 as a docker container.

Can anyone help me with what might be the error? I believe it is possibly that I do
[image]
[image]

n't have a target directory configured for digits on my P3 instance? The data looks correct to me.

AboutThatCMD, 2018-03-06:
@dusty-nv I performed a test and navigated to the efs/data level folder and went to n02799071 and the digits error is different. It says there must be at least two directories. This tells me it's reading the data. I am thinking about launching the Nvidia DIGITS 6 AMI and trying that approach tomorrow. Have you been able to successfully run through the steps using NVIDIA GPU Cloud and the Volta Deep Learning AMI?

AboutThatCMD, 2018-03-09:
@dusty-nv I was in touch with support from Nvidia and told that the symbolic links in the subset are the source of the problem with two classifications not being found.

dusty-nv, 2018-03-09:
In this case, you probably just want to make hard copies of the directories referenced in the script.  The symbolic links were used to save disk space and quickly be able to reconfigure the subclasses used by the 12 classification categories.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

AboutThatCMD, 2018-03-09:
@dusty-nv Thanks for the feedback. I will work on  doing that. This is my first project with this environment and it has been a great learning experience so far. If you have any advice on how to create the hard directories, that would be awesome. If not we can go ahead and close this out.

dusty-nv, 2018-03-09:
Basically you will want to replace the “ln -s” commands with cp commands (symbolic link vs copy)

AboutThatCMD, 2018-03-15:
Resolved by using the following modified script 

```
DATASET_PATH=$1
OUTPUT_PATH=$2

mkdir $OUTPUT_PATH/ball

cp -r $DATASET_PATH/n02799071 $OUTPUT_PATH/ball/baseball
cp -r $DATASET_PATH/n02802426 $OUTPUT_PATH/ball/basketball
cp -r $DATASET_PATH/n03134739 $OUTPUT_PATH/ball/croquet_ball
cp -r $DATASET_PATH/n04118538 $OUTPUT_PATH/ball/rugby_ball
cp -r $DATASET_PATH/n04254680 $OUTPUT_PATH/ball/soccer_ball
cp -r $DATASET_PATH/n04409515 $OUTPUT_PATH/ball/tennis_ball
cp -r $DATASET_PATH/n04540053 $OUTPUT_PATH/ball/volleyball

mkdir $OUTPUT_PATH/bear

cp -r $DATASET_PATH/n02132136 $OUTPUT_PATH/bear/brown_bear
cp -r $DATASET_PATH/n02133161 $OUTPUT_PATH/bear/black_bear
cp -r $DATASET_PATH/n02134084 $OUTPUT_PATH/bear/polar_bear
cp -r $DATASET_PATH/n02134418 $OUTPUT_PATH/bear/sloth_bear

mkdir $OUTPUT_PATH/bike

cp -r $DATASET_PATH/n02835271 $OUTPUT_PATH/bike/tandem_bicycle
cp -r $DATASET_PATH/n03791053 $OUTPUT_PATH/bike/scooter
cp -r $DATASET_PATH/n03792782 $OUTPUT_PATH/bike/mountain_bike

mkdir $OUTPUT_PATH/bird

cp -r $DATASET_PATH/n01514668 $OUTPUT_PATH/bird/rooster
cp -r $DATASET_PATH/n01514859 $OUTPUT_PATH/bird/hen
cp -r $DATASET_PATH/n01530575 $OUTPUT_PATH/bird/brambling
cp -r $DATASET_PATH/n01531178 $OUTPUT_PATH/bird/goldfinch
cp -r $DATASET_PATH/n01532829 $OUTPUT_PATH/bird/finch
cp -r $DATASET_PATH/n01534433 $OUTPUT_PATH/bird/junco
cp -r $DATASET_PATH/n01537544 $OUTPUT_PATH/bird/indigo
cp -r $DATASET_PATH/n01558993 $OUTPUT_PATH/bird/robin
cp -r $DATASET_PATH/n01560419 $OUTPUT_PATH/bird/bulbul
cp -r $DATASET_PATH/n01580077 $OUTPUT_PATH/bird/jay
cp -r $DATASET_PATH/n01582220 $OUTPUT_PATH/bird/magpie
cp -r $DATASET_PATH/n01592084 $OUTPUT_PATH/bird/chickadee
cp -r $DATASET_PATH/n01601694 $OUTPUT_PATH/bird/dipper
cp -r $DATASET_PATH/n01608432 $OUTPUT_PATH/bird/kite
cp -r $DATASET_PATH/n01614925 $OUTPUT_PATH/bird/bald_eagle
cp -r $DATASET_PATH/n01616318 $OUTPUT_PATH/bird/vulture
cp -r $DATASET_PATH/n01622779 $OUTPUT_PATH/bird/grey_owl

mkdir $OUTPUT_PATH/bottle

cp -r $DATASET_PATH/n02815834 $OUTPUT_PATH/bottle/beaker
cp -r $DATASET_PATH/n02823428 $OUTPUT_PATH/bottle/beer_bottle
cp -r $DATASET_PATH/n02823750 $OUTPUT_PATH/bottle/beer_glass
cp -r $DATASET_PATH/n03063599 $OUTPUT_PATH/bottle/coffee_mug
cp -r $DATASET_PATH/n04557648 $OUTPUT_PATH/bottle/water_bottle
cp -r $DATASET_PATH/n04560804 $OUTPUT_PATH/bottle/water_jug
cp -r $DATASET_PATH/n04591713 $OUTPUT_PATH/bottle/wine_bottle

mkdir $OUTPUT_PATH/cat

cp -r $DATASET_PATH/n02123045 $OUTPUT_PATH/cat/tabby
cp -r $DATASET_PATH/n02123159 $OUTPUT_PATH/cat/tiger_cat
cp -r $DATASET_PATH/n02123394 $OUTPUT_PATH/cat/Persian
cp -r $DATASET_PATH/n02123597 $OUTPUT_PATH/cat/Siamese
cp -r $DATASET_PATH/n02124075 $OUTPUT_PATH/cat/Egyptian
cp -r $DATASET_PATH/n02125311 $OUTPUT_PATH/cat/cougar
cp -r $DATASET_PATH/n02127052 $OUTPUT_PATH/cat/lynx
cp -r $DATASET_PATH/n02128385 $OUTPUT_PATH/cat/leopard
cp -r $DATASET_PATH/n02128757 $OUTPUT_PATH/cat/snow_leopard
[... 158 lines cut ...]
cp -r $DATASET_PATH/n01667778 $OUTPUT_PATH/turtle/terrapin
cp -r $DATASET_PATH/n01669191 $OUTPUT_PATH/turtle/box

mkdir $OUTPUT_PATH/vehicle

cp -r $DATASET_PATH/n02701002 $OUTPUT_PATH/vehicle/ambulance
cp -r $DATASET_PATH/n02930766 $OUTPUT_PATH/vehicle/taxi
cp -r $DATASET_PATH/n03345487 $OUTPUT_PATH/vehicle/fire_engine
cp -r $DATASET_PATH/n03417042 $OUTPUT_PATH/vehicle/garbage_truck
cp -r $DATASET_PATH/n03670208 $OUTPUT_PATH/vehicle/limousine
cp -r $DATASET_PATH/n03769881 $OUTPUT_PATH/vehicle/minibus
cp -r $DATASET_PATH/n03930630 $OUTPUT_PATH/vehicle/pickup_truck
cp -r $DATASET_PATH/n03977966 $OUTPUT_PATH/vehicle/police_wagon
cp -r $DATASET_PATH/n04065272 $OUTPUT_PATH/vehicle/rv
cp -r $DATASET_PATH/n03770679 $OUTPUT_PATH/vehicle/minivan
cp -r $DATASET_PATH/n04285008 $OUTPUT_PATH/vehicle/sports_car
cp -r $DATASET_PATH/n04461696 $OUTPUT_PATH/vehicle/tow_truck
cp -r $DATASET_PATH/n04467665 $OUTPUT_PATH/vehicle/tractor_trailer
cp -r $DATASET_PATH/n04487081 $OUTPUT_PATH/vehicle/trolleybus

mkdir $OUTPUT_PATH/sign

cp -r $DATASET_PATH/n06794110 $OUTPUT_PATH/sign/sign_street_sign
cp -r $DATASET_PATH/n06874185 $OUTPUT_PATH/sign/sign_traffic_light
```

closed 2018-03-15 by AboutThatCMD

msethi00, 2018-05-17:
The above solution worked for me too.

Vanderbeck, 2018-05-25:
Should add this fix to the [2 day tutoria.](https://github.com/dusty-nv/jetson-inference)
It still references the old code in github: [https://github.com/dusty-nv/jetson-inference/blob/master/tools/imagenet-subset.sh](https://github.com/dusty-nv/jetson-inference/blob/master/tools/imagenet-subset.sh)

eirikaso, 2018-06-29:
This worked for me to. 
Agree Vanderbeck. Should be added to the tutorial. 
It's quite frustrating when you start a two day tutorial and after three days you are only halfway due to a number of issues.

eirikaso, 2018-06-29:
That's good @dusty-nv 
I agree that links would have been best if it worked

hbe72, 2018-11-05:
If the ./imagenet-subset.sh script is run in the HOST side the symbolic link in the Digits docker container is dangling.

You have to follow the "Two Days to Demo" to the letter. The instructions in the "Two Days to Demo" state:  Run the folowing commands from the DIGITS server:
```bash
$ wget https://rawgit.com/dusty-nv/jetson-inference/master/tools/imagenet-subset.sh
$ chmod +x imagenet-subset.sh
$ mkdir 12_classes
$ ./imagenet-subset.sh /opt/datasets/imagenet/ilsvrc12 12_classes
```
You have to run the command from the DIGITS server (the running docker digits container), and within that you need to have write rights to the /data directory.

Therefore do not mount it the data directory in your docker run command as read only:
```bash
$ nvidia-docker run --name digits -d -p 8888:5000 \
 -v /home/username/data:/data
 -v /home/username/digits-jobs:/workspace/jobs nvcr.io/nvidia/digits:18.05
```
And then open interactive session to the container and run the imagenet-subset.sh script
```bash
$ docker exec -it digits bash
$ cd /data
$ ./imagenet-subset.sh /data/ilsvrc12 /data/12_classes
$ exit
```
As you have done this you can easily see that the created symbolic link is valid only at the Digits server side and the symbolic link is dangling in the host side.

sbow, 2019-05-12:
Ran into the same problem using Amazon AWS EC2 - NVIDIA-DIGITS-6.0 AMI
Solved it mostly the same way, with this slight variation:

AWS Shell host: sudo wget ... imagenet-subset.sh 
AWS Shell host: sudo chmod +x imagenet-subset.sh
AWS Shell host: sudo docker exec -it nvidia-digits bash
AWS Container side: cd data
AWS Container side: mkdir 12_classes
AWS Container side: ./imagenet-subset.sh /data/ilsvrc12 /data/12_classes/

vihari1729, 2019-11-06:
> Resolved by using the following modified script
>
> ```
> DATASET_PATH=$1
> OUTPUT_PATH=$2
>
>
> mkdir $OUTPUT_PATH/ball
>
> cp -r $DATASET_PATH/n02799071 $OUTPUT_PATH/ball/baseball
> cp -r $DATASET_PATH/n02802426 [...]

where are you modifying the script?

drakorg, 2020-05-05:
Hi, another possible way to go around this is to also make available to the docker containter the folder that the links reference, so that the exact same path referenced by the links is available on both sides. I went this way and I don't have to run the command everytime. You get this done with an extra -v /absolute_path_of_the_links_targets:/absolute_path_of_the_links_targets:ro ... (ro since the dataset is usually readonly).

Regards.