dusty-nv/jetson-inference #204: trying to compile imagenet-console.cpp

CrowMagnumMan, 2018-03-10:
I'm trying to compile imagenet-console.cpp as a starting point for writing some custom code based on it.  Is it possible to do that with a simple makefile?  I can't get it to work.  Thanks to anyone who can help.

Here's my makefile:
INC=-I/home/nvidia/jetson-inference \
	-I/home/nvidia/jetson-inference/build/aarch64/include \
	-I/usr/local/cuda-9.0/include

LIB=-L/home/nvidia/jetson-inference/build/aarch64/lib \
	-L/usr/local/cuda-9.0/lib64

all:
	g++ -std=c++11 $(INC) $(LIB) imagenet-console.cpp -o imagenet-consoleX

Here's the output of make:
/tmp/ccXxjo2n.o: In function `main':
imagenet-console.cpp:(.text+0x184): undefined reference to `imageNet::Create(int, char**)'
imagenet-console.cpp:(.text+0x1b0): undefined reference to `tensorNet::EnableProfiler()'
imagenet-console.cpp:(.text+0x1e8): undefined reference to `loadImageRGBA(char const*, float4**, float4**, int*, int*)'
imagenet-console.cpp:(.text+0x240): undefined reference to `imageNet::Classify(float*, unsigned int, unsigned int, float*)'
imagenet-console.cpp:(.text+0x2b4): undefined reference to `cudaFont::Create(char const*)'
imagenet-console.cpp:(.text+0x468): undefined reference to `cudaFont::RenderOverlay(float4*, float4*, unsigned int, unsigned int, char const*, int, int, float4 const&)'
imagenet-console.cpp:(.text+0x49c): undefined reference to `saveImageRGBA(char const*, float4*, int, int, float)'
imagenet-console.cpp:(.text+0x500): undefined reference to `cudaFreeHost'
/tmp/ccXxjo2n.o: In function `cudaCheckError(cudaError, char const*, char const*, int)':
imagenet-console.cpp:(.text._Z14cudaCheckError9cudaErrorPKcS1_i[_Z14cudaCheckError9cudaErrorPKcS1_i]+0x4c): undefined reference to `cudaGetErrorString'
collect2: error: ld returned 1 exit status
makefile:9: recipe for target 'all' failed
make: *** [all] Error 1

romzie, 2018-03-14:
I add a similar issue and I finally switched to cmake using the one from jetson-inference and removing the parts I did not want, and modifying paths, filenames and libraries. All my problems disppeared...

CrowMagnumMan, 2018-03-26:
Thanks for the tip. romzie.  Got that done the way you recommended and works fine.  I'd never done cmake before, so had to crash-course that.
You have to make sure that you edit CMakeLists.txt perfectly WRT where the include and util directories are, etc..   I was just messing with imagenet-console only so all I have in the project source directory is an empty build directory, the imagenet-console directory copied over from Jetson-inference, the CMakeLists.txt file, and the imageNet and tensorNet cpp, cu and h files.
Working on customizing my c++ around it now, Thanks!

CrowMagnumMan, 2018-03-29:
Working great, closing this issue.

closed 2018-03-29 by CrowMagnumMan

cameroncrose72, 2023-02-09:
@CrowMagnumMan @romzie, howdy!  I'm trying to do something similar.  I'm building a docker container based off of jetson-inference as well and I'm starting from stcratch.  My first goal is to successfully build and execute detectnet or imagenet or any of those projects in my own folder structure before I start developing the code for my project.  Can you include more information about which CMakeLists.txt file you edited and how you edited it?  For reference, here is my folder structure 
[image]
I have a top level folder with my Dockerfile, then a "container_src" folder with the "build", "inc", "src" folders and a CMakeLists.txt file.  Basically my intention is to use the "inc" and "src" folders for my project code, so I've put the "detectnet.cpp" in there, then I build my container (which runs CMake and Make) and then I am getting those same "undefined reference to" errors.  Any help you can offer would be appreciated!  I think the problem is related to the fact that my CMakeLists.txt doesn't know where to look for the code/object files where those functions are defined, and I don't know how to tell it where to look (if that's the issue).  Thank you in advance!

CrowMagnumMan, 2023-02-24:
@cameroncrose72  this one's 5 years out therefore pushing my brain...  If you're trying to write custom code, say based on detectnet.cpp, then these days with the latest jetson-inference updates I'd recommend this method to validate your stand-alone code (not involving docker).  Assuming that you've installed jetson-inference:
1. cd ~/ jetson-inference/examples
2. copy folder detectnet to a folder that is your custom name (e.g., MyCustomName)
3. cd /MyCustomName then change the CMakeLists.txt file to:
`file(GLOB detectnetSources *.cpp)
file(GLOB detectnetIncludes *.h )

cuda_add_executable(MyCustomName ${detectnetSources})
target_link_libraries(MyCustomName jetson-inference)
install(TARGETS MyCustomName DESTINATION bin)`
4. still within folder MyCustomName, rename detectnet.cpp to MyCustomName.cpp
5. add your custom code to MyCustomName.cpp
6. in terminal, cd  ~/jetson-inference/build
7. build everything in the build folder: cmake ../
8. compile everything:  make
9. cd ~/jetson-inference/build/aarch64/bin (or wherever your bin folder is)
10. run custom program:  ./MyCustomName