dusty-nv/jetson-inference #206: How to draw a diagnol line on the camera output

Pratosha-29, 2018-03-15:
I want to modify the code to draw a diagnol line on the camera output.
The changes need to be done in the file cudaOverlay.cu within the jetson-inference folder.
Can someone help me with a code to draw a diagnol line on the real time video. OpenCV and OpenGL methods tried do not work.
thanks.

dusty-nv, 2018-03-20:
Hi @Pratosha-29 , the line rasterization in `cudaOverlay` is very simple.  Right now it is setup for rectangles.  For each pixel, it simply checks if the x,y coordinates of that pixel are on the horizontal or vertical line of the rectangle.  This is achieved in this if statement in the code:   https://github.com/dusty-nv/jetson-inference/blob/e12e6e64365fed83e255800382e593bf7e1b1b1a/util/cuda/cudaOverlay.cu#L57

What you'll want to do is instead of checking against a rectangle, check each pixel's x,y values against the equation of a line (y=mx+b).  This will allow you to implement general line rasterization with lines of arbitrary slope (like diagonals).  Hope that helps!

Pratosha-29, 2018-03-21:
Hi @dusty-nv -
Thankyou. Yes it did help.

closed 2018-03-21 by Pratosha-29