dusty-nv/jetson-inference #207: getBindingIndex() returns different values when executing a shell script for SegNet model

jerisalan, 2018-03-15:
Hello there,

I am trying to execute the SegNet model on Jetson TX1. I have a test.sh file with the following contents:

```
NET=/home/nvidia/jetson-inference/20180302-022602-9d48_epoch_1.0

./segnet-console /home/nvidia/catkin_ws_1/src/beginner_tutorials/src/camera_image.png output_city.png \
--prototxt=$NET/deploy.prototxt \
--model=$NET/snapshot_iter_4522.caffemodel \
--labels=$NET/fpv-labels.txt \
--colors=$NET/fpv-deploy-colors.txt \
--input_blob=/home/nvidia/jetson-inference/data \
--output_blob=score_fr
```

When I execute the test.sh file from the terminal, the segnet model works perfectly. The problem arises when I try running the same with a python script inside a ROS package as I am trying to subscribe to a live feed and then trying to segment the image coming from the feed. The python script is executed using the following command.
```
rosrun beginner_tutorials pub_sub.py
```
Here is the python snippet in the pub_sub.py file that I use to call the shell script:
```
import os
os.system('./test,sh')
```

The only difference I see when the compare the outputs of both the approaches is that the getBindingIndex() function returns different values when the executable is run under different contexts.
I am posting where the difference arises:
Running test.sh from terminal
```
[GIE]  /home/nvidia/jetson-inference/20180302-022602-9d48_epoch_1.0/snapshot_iter_4522.caffemodel input  binding index:  0
[GIE]  /home/nvidia/jetson-inference/20180302-022602-9d48_epoch_1.0/snapshot_iter_4522.caffemodel input  dims (b=2 c=3 h=720 w=1280) size=22118400
[cuda]  cudaAllocMapped 22118400 bytes, CPU 0x100ce0000 GPU 0x100ce0000
```
Running test.sh from a python script within a ROS package
```
[GIE]  /home/nvidia/jetson-inference/20180302-022602-9d48_epoch_1.0/snapshot_iter_4522.caffemodel input  binding index:  -1
[GIE]  /home/nvidia/jetson-inference/20180302-022602-9d48_epoch_1.0/snapshot_iter_4522.caffemodel input  dims (b=2 c=0 h=0 w=0) size=0
failed to alloc CUDA mapped memory for tensorNet input, 0 bytes
```

Also to be noted is that the image is available before a call is made from the python script. Could anyone please point out why this odd behaviour is being observed.

dusty-nv, 2018-03-20:
Hi @jerisalan , are you able to launch any of the other programs from this repo using this launch method from ROS/python, like `imagenet-console` or `detectnet-console` for example?   It may be useful to see if these other programs work, with and without additional command line arguments, to narrow down the root cause.

jerisalan, 2018-03-21:
@dusty-nv Thanks for following up. Well I did not try running the other models in the same way as you mentioned to identify the root cause but I did get a workaround for the issue I was facing. 
I had to cd into the location where the segnet-console executable was located and then run the test.sh script. So the command in the python snippet had to be `os.system("cd /path/to/segnet-console-dir/ && ./test.sh")`.

closed 2018-04-25 by dusty-nv