dusty-nv/jetson-inference #208: Compiling using ROS

baroselfdriving, 2018-03-18:
Hi 

I have a problem when i'm using jetson-inference with ROS:

when i'm linking CXX Executable appear this error:

undefined reference to 'cv::namedWindow( 

and undefined reference to 'cv::imshow(

I'using jetson-inference , ROS Kinetic and OpenCV 3.0. 

Thanks!

dusty-nv, 2018-03-20:
Hi @baroselfdriving , are you linking openCV highgui library into your ROS application properly?   It seems like since you are getting undefined reference to functions related to GUI/windowing, it seems to point to missing highgui linkage.

baroselfdriving, 2018-03-20:
The problem is a core segmentation fault when i’m adding the jetson-inference library in my project.

proCoderDude, 2018-07-16:
Any updates to this issue? I am experiencing the same problem. The moment I link jetson-inference library to my ROS node executable, I get a segmentation fault.

closed 2023-04-27 by dusty-nv