dusty-nv/jetson-inference #213: nvinfer1::CudaError when attempting to run imagenet-console (Jetson TX2)

xoxota99, 2018-03-25:
After installing JetPack 3.2 (L4T 28.2) on a Jetson TX2, and following Instructions for "Building From Source on Jetson", when I tried to run the Image Classification example, "./imagenet-console orange_0.jpg output_0.jpg", I got the following cryptic error:
`imagenet-console
  args (3):  0 [./imagenet-console]  1 [orange_0.jpg]  2 [output_0.jpg]  

imageNet -- loading classification network model from:
         -- prototxt     networks/googlenet.prototxt
         -- model        networks/bvlc_googlenet.caffemodel
         -- class_labels networks/ilsvrc12_synset_words.txt
         -- input_blob   'data'
         -- output_blob  'prob'
         -- batch_size   2

[GIE]  TensorRT version 3.0, build 3004
[GIE]  attempting to open cache file networks/bvlc_googlenet.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  cudnnLayerUtils.cpp (288) - Cuda Error in smVersion: 38
terminate called after throwing an instance of 'nvinfer1::CudaError'
  what():  std::exception
Aborted (core dumped)`

Any ideas why? Did I do something wrong, and if so, what?

dusty-nv, 2018-04-25:
This is a strange error, because TX2 is `sm_62` (not `sm_38` like in the error above).

Are you sure the CUDA drivers were installed OK?   Can you run `deviceQuery` and `bandwidthTest` CUDA samples ok and post the output of `deviceQuery`?

superblr, 2018-07-27:
Hi, @xoxota99 @dusty-nv 
I have the same error.
but, I try [imagenet-camera]

hey, @dusty-nv
i installed cuda and CUDA sampled test OK.

Any ideas why?

closed 2023-04-27 by dusty-nv