dusty-nv/jetson-inference #214: CUDA error -  failed to build CUDA engine

groot-1313, 2018-03-26:
I am using an NVIDIA Jetson TX2, with Jetpack 3.1 and Tensorrt 3 installed. No other version of tensorrt exists on the TX2. 

Upon running `./imagenet-console orange_0.jpg output_0.jpg`, I get the following error:

> [GIE]  TensorRT version 3.0, build 3000
[GIE]  attempting to open cache file networks/bvlc_alexnet.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading networks/alexnet.prototxt networks/bvlc_alexnet.caffemodel
[GIE]  retrieved output tensor 'prob'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  cudnnEngine.cpp (55) - Cuda Error in initializeCommonContext: 4
[GIE]  failed to build CUDA engine
failed to load networks/bvlc_alexnet.caffemodel
failed to load networks/bvlc_alexnet.caffemodel
imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet

What needs to be done to resolve this?

flbeck, 2018-04-12:
I currently have the same issue, however on a PC with a Quadro M2000 GPU and TensorRT version 2.1

miunyhyr, 2018-05-22:
I currently have the same issue  on Jetson TX2 with JetPack 3.2. and TensorRT version 3.0.4

miunyhyr, 2018-05-22:
[GIE]  TensorRT version 3.0.4, build 3004
[GIE]  attempting to open cache file networks/bvlc_alexnet.caffemodel.2.tensorcache
[GIE]  cache file not found, profiling network model
[GIE]  platform has FP16 support.
[GIE]  loading networks/alexnet.prototxt networks/bvlc_alexnet.caffemodel
[GIE]  failed to retrieve tensor for output 'prob'
[GIE]  configuring CUDA engine
[GIE]  building CUDA engine
[GIE]  Network must have at least one input and one output
[GIE]  failed to build CUDA engine
failed to load networks/bvlc_alexnet.caffemodel
failed to load networks/bvlc_alexnet.caffemodel
imageNet -- failed to initialize.
imagenet-console:   failed to initialize imageNet

I am using an NVIDIA Jetson TX2, with Jetpack 3.2 and Tensorrt 3 installed. No other version of tensorrt exists on the TX2. And i have the same issue with you.

dusty-nv, 2018-05-22:
The error looks like corrupted alexnet.prototxt or alexnet.caffemodel, it cannot find the output layer "prob".  Van you try re-downloading and confirming the your prototxt has that layer name?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

daohu527, 2021-01-11:
Could you pls tell me some way to print out the debug message?
> [GIE] TensorRT version 3.0, build 3000
[GIE] attempting to open cache file networks/bvlc_alexnet.caffemodel.2.tensorcache
[GIE] cache file not found, profiling network model
[GIE] platform has FP16 support.
[GIE] loading networks/alexnet.prototxt networks/bvlc_alexnet.caffemodel
[GIE] retrieved output tensor 'prob'

dusty-nv, 2021-01-11:
Hi @daohu527 , you don't have an error in the log - does it load if you let it run for a few minutes?

Also it appears that you are running an older version of the project repo - if you are also running an older version of JetPack, that's ok, but if you are running a newer version of JetPack you may want to re-clone the project from master to get the latest updates.

closed 2023-04-27 by dusty-nv