dusty-nv/jetson-inference #218: Displaying image in imagenet-console.cpp

CrowMagnumMan, 2018-03-29:
imagenet-console.cpp saves the output image file to disk with the inference in text at the top of the picture using this:
saveImageRGBA(outputFilename, (float4*)imgCPU, imgWidth, imgHeight)

I want to, instead of saving to disk, throw the picture up on the screen.  This opencv command yields a blob of gray instead of the image:
cv::Mat image(cv::Size(imgWidth, imgHeight), CV_32FC4, (float4*)imgCPU )

I thought CV_32FC4 was the opencv equivalent to a float4.  I can't find anything on this.  Does anybody know how to do it?

CrowMagnumMan, 2018-04-02:
Never could figure this one out, so worked around it by using OpenCV to overlay the inference text on the original file and display that.

But this one still bugs me as it just seems ...... wrong.  So I'd like to keep this open for purely intellectual reasons.

dusty-nv, 2018-04-25:
Hi @CrowMagnumMan, it would be easier for you to take the openGL code from [`imagenet-camera`](https://github.com/dusty-nv/jetson-inference/blob/master/imagenet-camera/imagenet-camera.cpp) sample and insert it into imagenet-console.   It is done using the [`glDisplay`](https://github.com/dusty-nv/jetson-inference/blob/master/util/display/glDisplay.h) class.

closed 2018-04-25 by dusty-nv

dusty-nv, 2020-07-10:
FYI, for others who may see this - here is how convert to cv::Mat in C++:

https://github.com/dusty-nv/jetson-inference/issues/241#issuecomment-656773777