dusty-nv/jetson-inference #223: Different DetectNet output in DIGITS and JetsonTX2

eduardinjo, 2018-04-09:
I have trained custom object detection dataset fallowing this tutorial.
1. Originally 640x512 images -> re-scaled to 2560x2048
2. During training 640x512 image crops but full size in testing. Adjusted coverage threshold and min bounding box length for better detection.
3. Validation and Testing in DIGITS works GOOD.
4. For JetsonTX2 deleted clustering layer and adjusted threshold parameter with '--threshold=0.1' same as DIGITS. Using same image getting different bbox outputs. It seems those are the same detections but some just have bigger bounding box. 
           1. Are these the same outputs as DIGITS just clustered together similar to **eps** parameter setting? If so how to adjust it?
            2. #107 #48 #86 suggests issue with mean values, but no mean subtraction is performed from DIGITS interface. And detectnet layers have mean shift -127 for both train and deploy.  It seems this has been already [been taken care of.](https://devtalk.nvidia.com/default/topic/993552/jetson-tx1/detection-result-difference-between-jetson-inference2-3-and-digits5-1/post/5206253/#5206253)
            3. DIGITS (red), TX2 (blue)

@dusty-nv Could you kindly suggest possible cause of problem?

**Different question:** do I have to code pre-process scaling for images from 640x512 to 2560x2024 before using detectnet-camera or detectnet-console or this is taken care of automatically. Detectnet-console seemed to not scale them, but detectnet-camera had only 2FPS using default camera which is lower FPS than for default models.

[image]
[image]

bkanaki, 2018-04-09:
The clustering of bounding boxes is handled separately in the detectNet class [#L406](https://github.com/dusty-nv/jetson-inference/blob/e12e6e64365fed83e255800382e593bf7e1b1b1a/detectNet.cpp#L406). If you look at the merging logic closely, you can see that there is a very loose criterion for clustering the boxes unlike the one used on DIGITS. You can try to modify/reimplement this and this should solve your 'bigger box' issue. I have done the same because I was also dealing with similar issue.

As per the FPS, if you are displaying the 2560x2024 then a significant amount of computation will be used in rendering these images and thus leads to lower FPS. What is the network time reported on single image? This should be considered for the FPS of detection using the network, not considering the display or read/write time.

eduardinjo, 2018-04-11:
Hi. Thanks for clearing that up. I tried solution provided #195 which worked partially for me and hence re-implementing this is needed. What I understand from that thread is that confidence of each bbox is not taken into account during original clustering algo which can lead to worse accuracy depending on data. 
What did you use for your solution at the end? I'm thinking that maybe a non-max suppression algorithm might be appropriate but have to verify.

Inference speed was ~520ms for full image size so I should probably get ~1.9FPS if not rendering images. However I'll need to resize and save them on the fly so this should decrease FPS slightly.

closed 2018-04-11 by eduardinjo

kanakiyab, 2018-04-12:
I am using something that is very close to NMS. But it does slow down a bit the total inference time. I think that the GPU implementation for NMS can be done. But haven't worked on it yet.