dusty-nv/jetson-inference #224: Please add DetectNet-COCO- for  Cell Phone?

307509256, 2018-04-11:
I'm the beginners about  caffe&jetson.  

I want to detect Cell Phone，but the mAP of bvlc_googlenet or alexnet is unsatisfactory，  Can you give me the model of Cell Phone using COCO, such as DetectNet-COCO-CellPhone.tar.gz

I try to use digits, but is diffcult to me.

Thank you very much!  @dusty-nv

307509256, 2018-04-17:
maybe you can give us the url, above the whole caffemodel of COCO （80 class ) @dusty-nv

dusty-nv, 2018-04-25:
It appears that MS COCO has cell phone class:  [http://cocodataset.org/#explore](http://cocodataset.org/#explore)
Perhaps someone from the community has used it and can share their model.

307509256, 2018-04-25:
@dusty-nv
that will be diffcult to find it.
COCO have 80 classes, you offer 4 classes  only(Bottle, Dog,Airplane,Chair ). 
it's not enough.

closed 2023-04-27 by dusty-nv