dusty-nv/jetson-inference #233: Edge detection using opengl function

Raj7kumar7, 2018-05-15:
Hello
Is it possible to do edge detection on camera input using opengl.If yes,Can u share the code??

dusty-nv, 2018-05-15:
There are various OpenGL shaders implementing edge filters available if you search “GLSL edge detection”, like Sobel filter, Canny, ect.  However it is probably easier to integrate with this repo if using CUDA.   VisionWorks and NPP (NVIDIA Performance Primitives) have build-in functions for performing edge detection that use CUDA underneath.  Is there a particular reason that it needs to be OpenGL?

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

Raj7kumar7, 2018-05-15:
No ,not necessary Opengl . When using this repo,how can we integrate edge detection on camera input using VisionWorks and NPP ??Which  build-in functions will help in doing so??

closed 2023-04-27 by dusty-nv