dusty-nv/jetson-inference #234: Can you use the Jetson TX2 GPU for training new models on the host PC?

adam-gab, 2018-05-15:
Hello,

I am new to all of this and am going through this tutorial to get more familiar with all of these processes.

I am currently on the step "Creating Image Classification Model with DIGITS." When I followed the instructions directly, the training failed, saying it ran out of memory. When I looked this issue up it said I had to lower the batch size, which I switched to 5 from the default 10. It has now been running for over an hour without memory issues, but says it will take over 8 days to complete, which isn't optimal.

My computer has the 2GB Quadro M1000M graphics card, so I expect it to take longer than the example that used the 12GB Titan X and finished in less than 5 hours. Is this card so much slower that it would take 8 days to complete at best or is there a way to speed this up (does it sound like I'm doing something wrong)? Alternatively, is there a way to utilize the Jetson TX2 GPU that I will be using afterward for inferencing during this step that might make the process faster?

Any answers, tips or blunt honesty is greatly appreciated.

[image]

dusty-nv, 2018-05-15:
You could run less epochs (training iterations) on your M1000M, you don't necessarily have to wait for it to complete however many training epochs you had told it to run.

You can't control the TX2's integrated GPU from a PC, and DIGITS isn't officially supported on Jetson.

[screenshot]<https://user-images.githubusercontent.com/39314464/40079550-4432e36c-5856-11e8-8adb-b57a8d4908af.png>

-

-----------------------------------------------------------------------------------
This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

adam-gab, 2018-05-16:
Thanks for the quick response!

Would it be best if I wait for it to finish a whole number of epochs? After running overnight it is just about to finish #3.

Reading further into the tutorial, it looks like a model is saved after each epoch completes and the model is updated. As an example I don't need it to do too much, but will the model after epoch 3 likely have any functionality to it? In the example it looks like the model might change a lot through epoch 10.

If not, is there a way to resume training from where I abort the job? It seems like I could theoretically get 2-3 more epochs in each night while still using my computer during the day.

dusty-nv, 2018-05-16:
A model is saved by DIGITS after each epoch, so you probably want to keep the training running and test the models in the background.

It's possible to "resume" training by creating a new DIGITS model and specifying the old one as the Pre-Trained model.  The training will pick up where the old one left off.

adam-gab, 2018-05-16:
I am now trying to connect to the DIGITS server from the Jetson and it says "refused to connect" in the browser. I would like to download the most recent model onto the Jetson. Is this issue possibly due to the server currently training? Otherwise, I'm guessing it's due to the fact that I am on a public network right now and I might be screwed by the firewall.

dusty-nv, 2018-05-20:
@adam-gab hmm yes, sounds like a network or firewall issue.  I frequently connect to my DIGITS server from my Jetson's web browser (Chrome) on my LAN with no issues.

saracloud, 2020-01-11:
Not sure, it is good idea to raise this question quite old thread. 

A general question is, 
to run digits_server in a host PC for training, what is the PC configuration you would recommend.

closed 2023-04-27 by dusty-nv