dusty-nv/jetson-inference #238: How can I train segmentation using FCN-Alexnet woth 360p images?

teoac, 2018-05-27:
I'm following the examples about segmentation.
I don't have enough memory on my graphic card, so I want to train woth 360p images. (When I try training with 720p images, I always meet the 'out of memory' error.)
After training woth 360p images, the results  such as accuracy or loss, are very good and similar with tutorials.
But when I test using images, thr results are not good.
Is there any additional work to use 360p omages?

bpinaya, 2018-08-16:
What graphics card do you have? Are you using batches greater than 1? Maybe that's the issue. You shouldn't have problems to train with 720p

teoac, 2018-08-16:
@bpinaya Thank you for your reply. My graphics card is GTX 1060Ti w/ 3GB RAM. I cannot sure what batch size was set, but the size was larger than one.
As I said, the test result is very good on my host PC. But the test result on Jetson TX2 is poor as you can see above. I cannot understand why the result is not the same even if I used the same deep learning model.

bpinaya, 2018-08-16:
Uhm... don't worry, let's try to figure it out? What command are you using to deploy the network in the TX2? Maybe you are choosing a different network. also there will be some low resolution since Dustin is not using the deconvolution layer and just doing bilinear interpolation, even though the results should look better. From what I can see in your image it seems that the results are shifted since the slope seems to be the same.

closed 2023-04-27 by dusty-nv