dusty-nv/jetson-inference #240: Compiling using OpenCV

Pratosha-29, 2018-06-01:
Hi @dusty-nv ,
Added some lines to jetson-inference to perform lane detection .
Trying to compile jetson-inference , I am facing linker library issues.
I tried adding <include_directories(/usr/include/opencv2/)> in CMakeLists.txt
and even -lopencv in links.txt but failed.

Kindly help me resolve this issue.
These are the error logs faced.

[ 53%] Built target jetson-inference
[ 60%] Built target imagenet-console
[ 63%] Built target imagenet-camera
[ 68%] Built target detectnet-console
[ 70%] Linking CXX executable ../aarch64/bin/detectnet-camera
[ 75%] Built target segnet-console
[ 80%] Built target segnet-camera
[ 85%] Built target gst-camera
[ 90%] Built target v4l2-display
[ 95%] Built target v4l2-console
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `main':
detectnet-camera.cpp:(.text+0xaac): undefined reference to `cv::cvtColor(cv::_InputArray const&, cv::_OutputArray const&, int, int)'
detectnet-camera.cpp:(.text+0xb00): undefined reference to `cv::GaussianBlur(cv::_InputArray const&, cv::_OutputArray const&, cv::Size_<int>, double, double, int)'
detectnet-camera.cpp:(.text+0xb40): undefined reference to `cv::threshold(cv::_InputArray const&, cv::_OutputArray const&, double, double, int)'
detectnet-camera.cpp:(.text+0xb78): undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)'
detectnet-camera.cpp:(.text+0xbe8): undefined reference to `cv::Canny(cv::_InputArray const&, cv::_OutputArray const&, double, double, int, bool)'
detectnet-camera.cpp:(.text+0xc34): undefined reference to `cv::findContours(cv::_InputOutputArray const&, cv::_OutputArray const&, int, int, cv::Point_<int>)'
detectnet-camera.cpp:(.text+0xcbc): undefined reference to `cv::contourArea(cv::_InputArray const&, bool)'
detectnet-camera.cpp:(.text+0xd30): undefined reference to `cv::boundingRect(cv::_InputArray const&)'
detectnet-camera.cpp:(.text+0xd84): undefined reference to `cv::moments(cv::_InputArray const&, bool)'
detectnet-camera.cpp:(.text+0xf8c): undefined reference to `cv::drawContours(cv::_InputOutputArray const&, cv::_InputArray const&, int, cv::Scalar_<double> const&, int, int, cv::_InputArray const&, int, cv::Point_<int>)'
detectnet-camera.cpp:(.text+0xfec): undefined reference to `cv::rectangle(cv::Mat&, cv::Rect_<int>, cv::Scalar_<double> const&, int, int, int)'
detectnet-camera.cpp:(.text+0x1064): undefined reference to `cv::line(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
detectnet-camera.cpp:(.text+0x10ec): undefined reference to `cv::line(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
detectnet-camera.cpp:(.text+0x1170): undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)'
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `cv::String::String(char const*)':
detectnet-camera.cpp:(.text._ZN2cv6StringC2EPKc[_ZN2cv6StringC5EPKc]+0x40): undefined reference to `cv::String::allocate(unsigned long)'
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `cv::String::~String()':
detectnet-camera.cpp:(.text._ZN2cv6StringD2Ev[_ZN2cv6StringD5Ev]+0x10): undefined reference to `cv::String::deallocate()'
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `cv::Mat::Mat(int, int, int, cv::Scalar_<double> const&)':
detectnet-camera.cpp:(.text._ZN2cv3MatC2EiiiRKNS_7Scalar_IdEE[_ZN2cv3MatC5EiiiRKNS_7Scalar_IdEE]+0xb8): undefined reference to `cv::Mat::operator=(cv::Scalar_<double> const&)'
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `cv::Mat::~Mat()':
detectnet-camera.cpp:(.text._ZN2cv3MatD2Ev[_ZN2cv3MatD5Ev]+0x34): undefined reference to `cv::fastFree(void*)'
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `cv::Mat::create(int, int, int)':
detectnet-camera.cpp:(.text._ZN2cv3Mat6createEiii[_ZN2cv3Mat6createEiii]+0xd0): undefined reference to `cv::Mat::create(int, int const*, int)'
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `cv::Mat::release()':
detectnet-camera.cpp:(.text._ZN2cv3Mat7releaseEv[_ZN2cv3Mat7releaseEv]+0x58): undefined reference to `cv::Mat::deallocate()'
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `cv::Mat::operator()(cv::Rect_<int> const&) const':
detectnet-camera.cpp:(.text._ZNK2cv3MatclERKNS_5Rect_IiEE[_ZNK2cv3MatclERKNS_5Rect_IiEE]+0x24): undefined reference to `cv::Mat::Mat(cv::Mat const&, cv::Rect_<int> const&)'
CMakeFiles/detectnet-camera.dir/detectnet-camera.cpp.o: In function `void std::_Construct<cv::Moments>(cv::Moments*)':
detectnet-camera.cpp:(.text._ZSt10_ConstructIN2cv7MomentsEJEEvPT_DpOT0_[_ZSt10_ConstructIN2cv7MomentsEJEEvPT_DpOT0_]+0x34): undefined reference to `cv::Moments::Moments()'
collect2: error: ld returned 1 exit status
detectnet-camera/CMakeFiles/detectnet-camera.dir/build.make:127: recipe for target 'aarch64/bin/detectnet-camera' failed
make[2]: *** [aarch64/bin/detectnet-camera] Error 1
CMakeFiles/Makefile2:287: recipe for target 'detectnet-camera/CMakeFiles/detectnet-camera.dir/all' failed
make[1]: *** [detectnet-camera/CMakeFiles/detectnet-camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Thanks .

laksh-lp, 2019-03-30:
were you able to solve this ? even i wanted to convert this to mat image and was unable to do so ..

komms, 2019-07-20:
@Pratosha-29  Did you solve this problem of converting imgCPU to cv::Mat ?

dusty-nv, 2020-07-10:
See this post - https://github.com/dusty-nv/jetson-inference/issues/241#issuecomment-656773777

closed 2020-07-10 by dusty-nv