dusty-nv/jetson-inference #241: how convert imgRGBA late of cudaNormalizeRGBA(cuda processing)

hhamii2040, 2018-06-04:
hello 
I am trying to convert void* imgRGBA to OpenCV Mat Object
cv::Mat rgb_img = Mat(1080,1920,CV_8UC4 ,(unsigned char*)mRGBA);
Could anyone help me spot my bug?

CrowMagnumMan, 2018-06-05:
I was playing around with this earlier (issue #218).  I didn't have success making the conversion.  You could try using CV_32FC4 instead of CV_8UC4, and (float4*) instead of (unsigned char*).
I don't know why that didn't work for me, so I'd be really interested to see if that works with your code.

baroselfdriving, 2019-09-20:
Some solution for it?

dusty-nv, 2020-07-10:
Here is how to make it work:

https://forums.developer.nvidia.com/t/tx2-segmentation-fault/141327/14

```cpp
#include "cudaNormalize.h"  // include this above

CUDA(cudaNormalizeRGBA((float4*)imgRGBA,  make_float2(0,255),
					   (float4*)imgRGBA, make_float2(0,1),
					   imgWidth, imgHeight));
						 
CUDA(cudaDeviceSynchronize());
cv::Mat cv_image(cv::Size(imgWidth, imgHeight), CV_32FC4, imgRGBA);
cv::Mat cv_image2(cv::Size(imgWidth, imgHeight), CV_8UC3);
cv::cvtColor(cv_image, cv_image2, cv::COLOR_RGBA2BGR);
cv::imshow("Display window", cv_image2);
cv::waitKey(0);
```

manhtoanbkhn, 2020-12-22:
hello @dusty-nv , thank for your code. I tried it and get the error compile that cudaNormalizeRGBA function is not declared. Then I look up in the file cudaNormalize.h and change cudaNormalizeRGBA() -> cudaNormalize(), the project pass compile now. But when I run, there are an error segmentation fault (core dumped) at the line:
cv::cvtColor(cv_image, cv_image2, cv::COLOR_RGBA2BGR);
I cannot find out the solution to fix. Can you look at this please?