dusty-nv/jetson-inference #243: When I was about to give up...your repo gave me hope.

partnercloudsupport, 2018-06-06:
Thank you!

dusty-nv, 2018-06-06:
We're glad you enjoyed it, @partnercloudsupport.

partnercloudsupport, 2018-06-07:
@dusty-nv question for your Sir. Can I flash my Jetson from the EC2 instance?

Also...have you found a hard/fast way to get around the login loop issue once you download the cuda drivers?  It is the #1 issue I am having on my GPU's. Thank you in advance.

dusty-nv, 2018-06-07:
You can't flash your Jetson's remotely, they need to be attached directly to a native Ubuntu host PC.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

partnercloudsupport, 2018-06-07:
I thought so dusty....my heart is broken. Thank you. 

What is your experience with the login loop? If I go and buy a linux pc for this push...I am afraid I will keep getting locked out and have a brick. I just know you have a fix...you are a star.

dusty-nv, 2018-06-07:
Not sure what you mean login loop, do you mean on Jetson or a different discrete GPU?  On Jetson, use JetPack to install CUDA and the NVIDIA drivers.  They are not installed like normal.

partnercloudsupport, 2018-06-07:
Good question....on my ec2 p2/p3 gpu....I pull down nvidia drivers and then I do the nvidia-smi and everything looks good. Then I do a sudo reboot. 

I type my password in which is correct and it says password invalid. I have installed the drivers on 10+ gpu's and same thing every time. I unstall with sudo apt-get purge nvidia*  and walk through the steps described here https://gist.github.com/wangruohui/df039f0dc434d6486f5d4d098aa52d07 
and 1 out of 10 GPU's deployed it has worked on and unlock my box and I can work.

I have been working on this for a week and I am at my end....lol. I have to laugh to keep from crying. 
I hope this is clear. Thanks for responding.

closed 2023-04-27 by dusty-nv