dusty-nv/jetson-inference #246: Unable to create image classification model in DIGITS

eirikaso, 2018-06-29:
Hello 
I am trying to create the image classification model based on the 12 category subset. I have installed DIGITS using NGC.

Every time I try to make the image classification model this error occurs:

`Test net output #2: loss = 0 (* 1 = 0 loss)
Test net output #3: loss1/accuracy = 0
Test net output #4: loss1/accuracy-top5 = 0
Test net output #5: loss1/loss = 0 (* 0.3 = 0 loss)
Test net output #6: loss2/accuracy = 0
Test net output #7: loss2/accuracy-top5 = 0
Test net output #8: loss2/loss = 0 (* 0.3 = 0 loss)
Initial Test completed in 0.140375s
Restarting 4 internal thread(s) on device 0
Starting 1 internal thread(s) on device 0
Data Reader threads: 3, out queues: 12, depth: 32
Starting 3 internal thread(s) on device 0
NVML succeeded to set CPU affinity on device 0, thread 167
Opened lmdb /workspace/jobs/20180629-115535-bdb4/train_db
Opened lmdb /workspace/jobs/20180629-115535-bdb4/train_db
Output data size: 32, 3, 224, 224
Parser threads: 3 (auto)
Transformer threads: 4 (auto)
Opened lmdb /workspace/jobs/20180629-115535-bdb4/train_db
Check failed: error == cudaSuccess (48 vs. 0)  no kernel image is available for execution on the device
        
`

I tried to "downgrade" my JetPack installation to 3.1 in order to use cuda 8 instead of 9, but got the same error. 
My computer has a Quadro M1200 GPU.

dusty-nv, 2018-06-29:
`no kernel image is available for execution on the device` means that the SM profile that the code was compiled with doesn't support the compute capability of your GPU card.  In your case, I would recommend ignoring JetPack on the host (which is only really there to flash/setup your Jetson with the latest sw) and instead install CUDA toolkit + NV driver on your PC from Ubuntu apt or from NVIDIA .run file.  If you continue having problems with your NGC/Docker setup, you may wish to just install DIGITS natively like the one path from the tutorial.

eirikaso, 2018-06-29:
Thank you for your quick reply. I will try  to uninstall JetPack again. 

One additional question about the NGC. As I understand it this can be used to rent external  GPU processing capacity. Is this correct? Will I be able to do this if I don't install DIGITS through NGC?

dusty-nv, 2018-06-29:
NGC provides containers which are compatible with cloud providers like AWS and MSFT, which will rent you the GPUs.  There are several containers for more than just DIGITS, like TensorFlow or vanilla Ubuntu, ect.

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

closed 2019-06-21 by dusty-nv