dusty-nv/jetson-inference #252: Make error on V100, "/usr/bin/ld: cannot find -lnvinfer"

wuqi930907, 2018-07-31:
Hi,
I want make jetson-inference on V100, but something seems error:
/usr/bin/ld: cannot find -lnvcaffe_parser
/usr/bin/ld: cannot find -lnvinfer
collect2: error: ld returned 1 exit status
CMakeFiles/jetson-inference.dir/build.make:1723: recipe for target 'x86_64/lib/libjetson-inference.so' failed
make[2]: *** [x86_64/lib/libjetson-inference.so] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

But libnvcaffe_parser.so and libnvinfer.so are in $LD_LIBRARY_PATH(I used TensorRT-2.1.2.x86_64.cuda-8.0-16-04 and TensorRT-3.0.4.Ubuntu-16.04.3.x86_64.cuda-9.0.cudnn7.0 to test).
Is jetson-inference can only make success on TX2?

Thanks

closed 2018-08-03 by wuqi930907

vinitmuchhala, 2019-04-02:
Hey, I am getting the same error, is there a fix for this that you figured out?

akashzade, 2021-07-21:
I am also getting like same error while make

`g++ -c -o nvdsparsebbox_Yolo.o -Wall -std=c++11 -shared -fPIC -Wno-error=deprecated-declarations -I../../includes -I/usr/local/cuda-11.1/include -I/home/ubuntu/Downloads/softwares/tensorrt/TensorRT-7.2.3.4/include nvdsparsebbox_Yolo.cpp
g++ -o libnvdsinfer_custom_impl_Yolo.so  nvdsparsebbox_Yolo.o -shared -Wl,--start-group -lnvinfer_plugin -lnvinfer -lnvparsers -L/usr/local/cuda-11.1/lib64 -lcudart -lcublas -lstdc++fs -Wl,--end-group
/usr/bin/ld: cannot find -lnvinfer_plugin
/usr/bin/ld: cannot find -lnvinfer
/usr/bin/ld: cannot find -lnvparsers
collect2: error: ld returned 1 exit status
Makefile:48: recipe for target 'libnvdsinfer_custom_impl_Yolo.so' failed
make: *** [libnvdsinfer_custom_impl_Yolo.so] Error 1
`

dusty-nv, 2021-07-21:
@akashzade that code is not from this repo, please post to the DeepStream forum -

https://forums.developer.nvidia.com/c/accelerated-computing/intelligent-video-analytics/deepstream-sdk/15