dusty-nv/jetson-inference #263: Low accuracy inference outputs...

eanmikale, 2018-09-07:
@dusty-nv - Also having issue with training model. Accuracy is in 80's as you see below. Training on four Titan V's. Batch sizes is 4, and validation percentage is 10%, but have also tried 25%, with no changes. Accuracy level for top five inferences are as follows (3.08%, 3.08%, 3.07%, 3.07%, and 3.07%). We have cleaned our jobs folder and tried again, as well as rebuilding DIGITS, but are still getting these results. Any assistance would be appreciated. Thanks again. 

[image]

perronj2, 2019-01-30:
I'm having the same issue, I believe it could be due to a significant amount of broken links in imagenet. The decrease in sample size will decrease accuracy of the inference... The links really need to be updated.

perronj2, 2019-01-30:
@dusty-nv I have been waiting approval for close to a year....

dusty-nv, 2019-01-30:
I was able to find a copy on Academic Torrents (for your research purposes only)

This email message is for the sole use of the intended recipient(s) and may contain
confidential information.  Any unauthorized review, use, disclosure or distribution
is prohibited.  If you are not the intended recipient, please contact the sender by
reply email and destroy all copies of the original message.
-----------------------------------------------------------------------------------

perronj2, 2019-01-30:
@dusty-nv Thanks for the info!

closed 2023-04-27 by dusty-nv